devWaves/SwitchBot-MQTT-BLE-ESP32 #2: Turns on at random?

hgelpke, 2021-04-10:
I got this all setup and it works but the one issue is that the ESP randomly sends the  command to the SB for a button push.  No external triggers are telling it to do so as far as I can tell. 

Might be related to loss/reconnect of wifi but I'm not certain

devWaves, 2021-04-10:
can't say that really makes sense based on the way it is coded. It only calls the command once if u send it. Are u republishing on reboot or something?

I cannot reproduce that issue. I only own buttons

What MQTT broker are u using and what is publishing/subscribing the MQTT?

devWaves, 2021-04-11:
if you are using HA, another user had a similar issue with mosquitto mqtt 5.1.1, they reverted back to their older version and it fixed this issue

this link has workaround if that is the exact issue...https://community.home-assistant.io/t/mosquitto-5-1-1add-on-is-broken/286979/21

hgelpke, 2021-04-11:
Thank you

I've re-flashed and it seems good now.

Also may have been the case I printed causing issues. 

Wish I could say 100% but I appreciate the work you're doing and the quick response.

devWaves, 2021-04-11:
no problem. I'll close the issue for now. can be reopened if needed. im glad others have found this code to be useful

closed 2021-04-11 by devWaves