devWaves/SwitchBot-MQTT-BLE-ESP32 #5: Slow reaction time every now and then

Pierre-33, 2021-05-10:
First, many thanks for this great project, I'm doing my first Arduino step with it :)

I only use the push command on a SwitchBot and it works like a charm most of the time, but every now and then it takes a very long time (something like a minute) between the MQTT command and the actual reaction of the SwitchBot.

The SwitchBot is literally next to the esp32, so it shouldn't be related to Bluetooth range. 

It's my first time using MQTT as well so I can't be sure if this delay is caused by the broker (I'm using the one provided with Home Assistant) or by the esp32, but if I would guess, I would say a very long process is triggered every now and then on the esp32 which makes it unresponsive for a few minutes.

Is that a know behavior?

devWaves, 2021-05-11:
if you also have the actual switchbot hub, I can't say if that causes any issues or not. It is possible a switchbot hub could lock the bot temporarily while connected for some reason, not allowing other connections. But I have no idea

Pierre-33, 2021-05-11:
Thanks for your quick answer! :)

>if you are only using the bots (like me) make sure the meter and curtain map lists are empty or the inside is commented out (otherwise the scan will be looking forever for a non existing MAC). Make sure the bot list only has valid Mac addresses

That should be it! I replace switchbotone with my switch bot and left switchbottwo with it's fake Mac address, I'll try that asap, but I have good hope it should fix my issue :)

>in version 1.5 you will now see the onboard LED on the esp32 turn on while it is busy (with the default settings). If the LED is on, that means the esp32 is busy. If the LED is not on, a control command should be somewhat instant. (not the same as the power LED, that is always on)

That should be really helpful. On a side note, it would be really helpful as well to have the configuration in a separate file (like a .h") if you plan to release regular update :)

closed 2021-05-11 by devWaves