devWaves/SwitchBot-MQTT-BLE-ESP32 #7: LED_BUILTIN Error

marithpl, 2021-05-17:
Im using version 0.14 today I received second switch curtain and by the way I wish to update, but when I compile I've got this error in Arduino:
```
In function 'void scanEndedCB(NimBLEScanResults)':
sketch_may17a:484:18: error: 'LED_BUILTIN' was not declared in this scope
     digitalWrite(LED_BUILTIN, LOW);
```

devWaves, 2021-05-17:
0.15 is the latest stable version and for LED stuff. 0.2 is being wotked on for HA MQTT Discovery

the LED_BUILTIN is defined by default on some ESP32, sounds like you mught have one that doesn't. But I noticed you were testing the HA MQTT discovery stuff? did you remove the LED code, because it is there also

on my D1 Mini ESP32 LED_BUILTIN= 2

I will add a fix at some point. But you will need to determine what pin is the onboard led pin. You can simply replace LED_BUILTIN everywhere with the led pin number, or create a variable example: int LED_PIN = 2

I wad trying to avoid another config value but it looks like it might be needed for certain esp32 boards

marithpl, 2021-05-17:
I've added `int LED_BUILTIN = 2;` and it works.
Also I ave to add it to MQTT Discovery stuff because I had the same error.

closed 2021-05-17 by marithpl