devWaves/SwitchBot-MQTT-BLE-ESP32 #8: Example switchbot mqtt config

robbinonline, 2021-05-19:
Thanks for this script, I got it working on a ESP.
Can you give a example how to add a switch into home assistant?

What is the command_topic and state_topic when I add a mqtt switch: https://www.home-assistant.io/integrations/switch.mqtt/

and what to add by state_on / off when it only is a press?

charlyschulte, 2021-05-19:
i would suggest waiting a bit. there are working on an mqtt home assistant auto discovery feature.
Then you wont need to integrate it manually. Check the pull request out here: https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/pull/6

you can also use the alpha version from @HardcoreWR: https://github.com/HardcoreWR/SwitchBot-MQTT-BLE-ESP32

Schluggi, 2021-05-20:
Until the autodiscover feature is ready to use, you can use my mqtt config as an example.
Tested with v1.4 but should also work with v1.5.

```yaml
sensor:
  - platform: mqtt
    name: curtain_bedroom_left
    state_topic: "switchbot/curtain/curtain_bedroom_left"
    value_template: "{{ 100 - (value_json.pos | int) }}" 
    json_attributes_topic: "switchbot/curtain/curtain_bedroom_left"

  - platform: mqtt
    name: curtain_bedroom_right
    state_topic: "switchbot/curtain/curtain_bedroom_right"
    value_template: "{{ 100 - (value_json.pos | int) }}"
    json_attributes_topic: "switchbot/curtain/curtain_bedroom_right"

cover:
  - platform: template
    covers:
      bedroom_left:
        device_class: cover
        friendly_name: Schlafzimmervorhang links
        position_template: "{{ states('sensor.curtain_bedroom_left') }}"
        set_cover_position:
          service: mqtt.publish
          data:
            topic: switchbot/control
            payload_template: '{ "id": "curtain_bedroom_left", "value": "{{ position }}" }'
        open_cover:
          service: mqtt.publish
          data:
            topic: switchbot/control
            payload: '{ "id": "curtain_bedroom_left", "value": "open" }'
        close_cover:
          service: mqtt.publish
          data:
            topic: switchbot/control
            payload: '{ "id": "curtain_bedroom_left", "value": "close" }'
        stop_cover:
          service: mqtt.publish
          data:
            topic: switchbot/control
            payload: '{ "id": "curtain_bedroom_left", "value": "pause" }'
        icon_template: >-
          {% if states('sensor.curtain_bedroom_left') | int > 10 %}
            mdi:window-shutter-open
          {% else %}
            mdi:window-shutter
          {% endif %}

      bedroom_right:
        device_class: cover 
        friendly_name: Schlafzimmervorhang rechts
        position_template: "{{ states('sensor.curtain_bedroom_right') }}"
        set_cover_position:
          service: mqtt.publish
          data:
            topic: switchbot/control
            payload_template: '{ "id": "curtain_bedroom_right", "value": "{{ position }}" }'
        open_cover:
          service: mqtt.publish
          data:
            topic: switchbot/control
            payload: '{ "id": "curtain_bedroom_right", "value": "open" }'
        close_cover:
          service: mqtt.publish
          data:
            topic: switchbot/control
            payload: '{ "id": "curtain_bedroom_right", "value": "close" }'
        stop_cover:
          service: mqtt.publish
          data:
            topic: switchbot/control
            payload: '{ "id": "curtain_bedroom_right", "value": "pause" }'
        icon_template: >-
          {% if states('sensor.curtain_bedroom_right') | int > 10 %}
            mdi:window-shutter-open
          {% else %}
            mdi:window-shutter
          {% endif %}
```

closed 2021-05-20 by devWaves