devWaves/SwitchBot-MQTT-BLE-ESP32 #15: sdkconfig.h: No such file or directory

npirtube, 2021-06-14:
Hello, good morning,

First of all I would like to thank you for the work that you have done, congratulations!

On the other hand, I am trying to install on a Wemos D1 Mini board through the Arduino IDE.

But I have encountered the following error.

Arduino:1.8.15 (Windows 10), Tarjeta:"WeMos D1 R1, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

In file included from D:\Documents\Home Assistant\SwitchBot\SwitchBot-MQTT-BLE-ESP32-main\SwitchBot-BLE2MQTT-ESP32\SwitchBot-BLE2MQTT-ESP32.ino:258:0:

NimBLEDevice.h:17:23: fatal error: sdkconfig.h: No such file or directory

 #include "sdkconfig.h"

                       ^

compilation terminated.

exit status 1

sdkconfig.h: No such file or directory

Please, Can anyone tell me how I can solve it?

devWaves, 2021-06-14:
did you setup your IDE for the esp32 type you have? and then install all the required libraries?

the error your getting is most likely an IDE setup issue. I think that file is generated by the IDE

This is an issue many user have had, you shouldnt have to manually copy that file

maybe try running the IDE as admin?

It is possible one of the required libraries was updated recently and broke, but not that I am aware of

npirtube, 2021-06-14:
_Did you configure your IDE for the type of esp32 you have, and then install all the necessary libraries?_
Yes, I can load other projects without problems.

_Are you trying to run the IDE as administrator?_
I have also tried, but without success

I have made sure I have all the libraries up to date. I have loaded other test sketches correctly.
But every time I try to load this project, I always get the same error.
Also, if I generate the sdkconfig.h file manually, then it tells me that a different file is missing.

I will continue testing.

npirtube, 2021-06-14:
I have continued testing and finally the error was in the IDE configuration, the board control was not configured for an ESP32.

I was able to successfully integrate with my HA installation.
Thank you very much for your work.

devWaves, 2021-06-14:
no problem. glad it is working for you

closed 2021-06-14 by devWaves