devWaves/SwitchBot-MQTT-BLE-ESP32 #20: Support motion detection

cpeuschel, 2021-07-20:
Hi, 

is it possible to support motion detection sensors?

cpeuschel, 2021-07-20:
i have tried to implement an motion sensor.
On your code is an condition `if (advertisedDevice->isAdvertisingService(NimBLEUUID("cba20d00-224d-11e6-9fb8-0002a5d5c51b")))` but the new Motion Device has no serviceUUID. Any ideas?

devWaves, 2021-07-20:
I haven't looked at how the switchbot motion detectors work yet. The code would need some big changes specific for motion though I think

motion detection would need a constant connection to the device, or at least something scanning constantly for BLE messages it sends... im guessing

the current code only scans and connects when needed. 

If motion needs constant connections the esp32 won't work. If the motion detector only broadcasts once, that will be a problem. If the motion detector is always broadcasting lastMotionTimestamp, then we could capture that

it might not be a super fast motion response time though

I am going to close this issue for now. feel free to continue looking into it though

closed 2021-07-20 by devWaves

devWaves, 2021-09-04:
I have started to look into the motion sensor and contact sensor. You can follow the status from this ticket
https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/issues/27

I started a new branch for testing purposes

devWaves, 2021-09-19:
motion sensor and contact sensor support added in v6.0