devWaves/SwitchBot-MQTT-BLE-ESP32 #21: Problem with removed devices

marithpl, 2021-07-25:
I’ve got a problem with showing switchbot devices in HA. 

By mistake I removed a switchbot devices from HA. I changed a config of SwitchBot MQTT but still there is no devices showing in HA. 

How to “reset” ESP to show devices in HA?

devWaves, 2021-07-25:
when the esp32 reboots it sends the mqtt discovery info. your devices will automatically create when the esp32 reboots. if it does not that is an issue with your HA.

ensure the device is fully deleted, then reboot the esp32

closed 2021-07-25 by devWaves

marithpl, 2021-07-26:
How to ensure the device is fully deleted?

devWaves, 2021-07-26:
Configurations->Devices

if the switchbot is in the list, click it and select delete

marithpl, 2021-07-26:
I've deleted this way.
And now MQTT do not discovery again switchbot's.

I tried to remove MQTT integration an add a new one. My tasmotas devices showed up again but switchbot not.

In MQTT Explorer I can find switchbots curtains:
[image]

devWaves, 2021-07-26:
hmm... not sure
I've done it many many times, no issues. but with bots. it should work the same for curtain though.
let me google a little bit...

devWaves, 2021-07-26:
can u try with short names? maybe the discovery msgs are failing with long names. just for testing, make the names short and see if devices are discovered

devWaves, 2021-07-26:
oh... you have spaces in the names. I think thats it

marithpl, 2021-07-26:
I will check in the morning

marithpl, 2021-07-27:
Now is working. 

Thanks!