devWaves/SwitchBot-MQTT-BLE-ESP32 #22: Support for Humidifier

charlyschulte, 2021-07-26:
Is it possible to use the SwitchBot Humidifier
(https://www.switch-bot.com/products/switchbot-smart-humidifier)
or is it possible to integrate the support?

would be really great

closed 2021-07-26 by devWaves

DirkTas67, 2021-11-17:
I'm interested in this too, maybe @devWaves can find info in this github repo
[switchbotapi](https://github.com/OpenWonderLabs/SwitchBotAPI)

devWaves, 2021-11-17:
There is currently no BLE documentation for the Humidifier.

The api documentation is for cloud stuff, not BLE so it doesn't help

I will try to continue looking into this, but without documentation it will be guessing and reverse engineering, so no promises

With proper documentation it would be fairly easy to implement

If you confirm you can control the humidifier with BLE only then that helps.

To test this, turn your phone WIFI off. Unplug switchbot hub if you have one. Turn bluetooth on your phone on, then try controlling the humidifier from the switchbot app

if you can control it with BLE, then there is potential of supporting it