devWaves/SwitchBot-MQTT-BLE-ESP32 #25: Add MQTT channel/attribute for wifi rssi

gdolfen, 2021-08-17:
Can you please add a MQTT channel or an attribute to an existing channel for wifi rssi.

Here is a code snipped:

   String rssi = String(WiFi.RSSI());
   client.publish("switchbot/esp32/rssi", rssi);

devWaves, 2021-08-17:
ya sure no problem. I'm looking at adding some updates for low battery, I will add it in that release

before I get to it, you can try adding your code snippet in this method that gets called every 30 sec

```
void publishLastwillOnline() {
  if ((millis() - lastOnlinePublished) > 30000) {
    if (client.isConnected()) {
      client.publish(lastWill, "online", true);
      lastOnlinePublished = millis();
    }
  }
}
```

closed 2021-08-17 by devWaves

devWaves, 2021-09-19:
v6.0 adds an esp32 device to HA with rssi value