devWaves/SwitchBot-MQTT-BLE-ESP32 #26: Speed of Bot and esp32

spartandrew18, 2021-08-23:
Is there a way to increase at which the bot responds to the esp32? With the app and phone it's pretty fast but with home assistant, there is a couple of seconds delay. Is there somewhere in the code I could modify the time it takes? Thank you!

devWaves, 2021-08-23:
it all depends on the bluetooth signal between the bot and the esp32. the esp32 should be close to the bot if you want it to work instantly

what is the rssi value you get in the attributes?

my main bot is very close to the esp32 and it works basically instantly

spartandrew18, 2021-08-23:
Yes, my bot and the esp 32 are right next to my room in the bathroom. It is not as fast as the app. How do I find the RSSI value?

devWaves, 2021-08-23:
it is published in mqtt attributes for the device.
<ESPMQTTTopic>/bot/<name>/attributes

main github page has that info

it may depend on the esp32. I can say that mine work almost instantly. there are some very minor delays in the code for stability, but if you are experiencing seconds delays it means it is just taking a long time to connect

where are you standing when using the app? I just tried with the app (using an S9 samsung) while standing in my dining room, bot in the kitchen, it did 3/4 of a circle....so its not instant with the app for me

spartandrew18, 2021-08-23:
[image]

I am in home assistant. Do I run something here to listen to the device?

devWaves, 2021-08-23:
are u using MQTT discovery? if so open the device. it will be listed as LinkQuality

spartandrew18, 2021-08-23:
[image]

Just this. I don't quite understand most of this stuff honestly. I just followed your guide to upload the software from Arduino IDE and ran the program.

devWaves, 2021-08-23:
in HA main dashboard, go to Configurations, then devices. The switchbot device will be listed if you are using mqtt discovery

open that and you should see LinkQuality with an rssi number. 

I cant send screenshots atm

spartandrew18, 2021-08-23:
The device is not here. I only have an entity for the device. Am I supposed to do something in the config.yaml folders?

[image]

devWaves, 2021-08-23:
you should have to do zero config changes in HA. no yaml files to create/edit when using mqtt discovery. the devices automatically show up

rebooting the esp32 resends the data for HA to automatically find it

closed 2021-08-23 by devWaves

devWaves, 2021-08-24:
Did you manually create that device? that must be one of the other switchbot integrations? I am assume you tried to get those working.  Are you sure you are even calling the esp32? 

These are the HA steps to follow (if you setup the ESP32 correctly)

[image]
[image]
[image]
[image]

spartandrew18, 2021-08-25:
So I just reset my whole home assistant. This is my flashed setup in IDE. Is this setup correctly for the esp32?

/****************** CONFIGURATIONS TO CHANGE *******************/

/********** REQUIRED SETTINGS TO CHANGE **********/

/* Wifi Settings */
static const char* host = "esp32switchbothub";                          //  Unique name for ESP32. The name detected by your router and MQTT. If you are using more then 1 ESPs to control different switchbots be sure to use unique hostnames. Host is the MQTT Client name and is used in MQTT topics
static const char* ssid = "Anderson 2.4G";                           //  WIFI SSID
static const char* password = "mypassword";                   //  WIFI Password

/* MQTT Settings */
/* MQTT Client name is set to WIFI host from Wifi Settings*/
static const char* mqtt_host = "10.0.0.207";                       //  MQTT Broker server ip
static const char* mqtt_user = "mqtt";                         //  MQTT Broker username. If empty or NULL, no authentication will be used
static const char* mqtt_pass = "mqtt";                         //  MQTT Broker password
static const int mqtt_port = 1883;                                  //  MQTT Port
static std::string mqtt_main_topic = "switchbot";                   //  MQTT main topic

/* Switchbot Bot Settings */
static std::map<std::string, std::string> allBots = {
  /*{ "switchbotone", "d3:15:04:54:18:c0" },
    { "switchbottwo", "yY:yY:yY:yY:yY:yY" }*/
};

I am not seeing any autodiscovery happening in mosquitto broker.

devWaves, 2021-08-25:
you have a typo here.....

```
/* Switchbot Bot Settings /
static std::map<std::string, std::string> allBots = {
/{ "switchbotone", "d3:15:04:54:18:c0" },
{ "switchbottwo", "yY:yY:yY:yY:yY:yY" }*/
};
```

should be this
```
/* Switchbot Bot Settings /
static std::map<std::string, std::string> allBots = {
{ "switchbotone", "d3:15:04:54:18:c0" }
};
```

spartandrew18, 2021-08-25:
THANK THE LORD! AND THANK YOU!

[image]
[image]
[image]

devWaves, 2021-08-25:
awesome!

-72 will work fine, but ya it could be better. My printer switchbot is between 70-80. It works fine but can take a bit. My coffee switchbot is in the 50s

if you can get a signal in the 50s or below then its basically instantaneous

they do make esp32s with external antennas also