devWaves/SwitchBot-MQTT-BLE-ESP32 #27: adding Contact Sensor & Motion Sensor

BangerTech, 2021-09-02:
Hi there, 

first of all a huuuge Thank you for the great work!!!

i was wondering if ther would/ will be a way to connect the contact & motion sensor in the near future?

thanks again 

greets

devWaves, 2021-09-02:
Hi. This has already been requested for motion sensor here... https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/issues/20

switchbot has not released documentation on how the contact or motion sensor work, so without that there isn't much I can do

My guess is the current solution would not work, especially if the devices only broadcast the "motion" "no motion" "open" "closed" message once

the esp32 bluetooth cannot scan for devices while at the same time connecting to control bots. If you try to control a bot, and during that time motion or open is broadcasted, the esp32 would miss the broadcast.

if switchbot releases documentation I'll take a look. It may work if the esp32 is only used for motion sensors, contact sensors, and temp meters because they don't require a connection. My guess is that combining all the switchbot device types on one esp32 won't work.

Maybe the switchbot hub has more than one bluetooth radio and that's how they do it, I have no idea

at the moment I can only make guesses on how they work

devWaves, 2021-09-02:
ya I think I can make it work, but it will take some work. But while you are controlling a bot, you won't get motion/contact updates. I will need to make the code scan all the time, but not send MQTT updates all the time

problem is I don't have either. Do u have both? I will for sure need a tester

BangerTech, 2021-09-02:
that sounds great! I think i wouldn’t be a problem for the users to wait till they are finished controlling a bot right? ;-) 
yes i got both so would be no problem to be your tester ;-)

devWaves, 2021-09-02:
waiting shouldn't be a big problem ... I think. I feel like all implementations would work that way

within the switchbot app how long does it say motion for? what is the motion timeout

I noticed the contact sensor has a button on it also. That could be tricky to integrate

BangerTech, 2021-09-03:
soo i would say let´s start with the contact sensor. Basically the contact sensor reading would be enough for me. But on this Contact Sensor there is a Light Indicator / Sensor, a Fresnel Lens / Pir Sensor and a Reset Button.

devWaves, 2021-09-04:
I have created a new branch for testing the motion sensor and contact sensor
https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/v6MotionContact/SwitchBot-BLE2MQTT-ESP32.ino

Please try this code out and let me know if you are getting information from the contact sensors and motion sensors and let me know if the information looks correct

Currently in the code a contact sensor and motion sensor will have their attributes collected the same was as meters. Every 30 secs a scan will be made. **This is only temporary**. I will make the code changes to constantly scan after I can confirm the values are at least correct

These are the values I am collecting....

**For motion sensor**
- battery
- linkquality
- motion = possible values : "MOTION", "NO MOTION"
- illuminance = possible values : "BRIGHT", "DARK"
- led = possible values : "ON", "OFF"
- sensedistance = possible values : "LONG", "MIDDLE", "SHORT"
- lastmotion = NOT COMPLETED, will eventually be a timestamp

**For contact sensor**
- battery
- linkquality
- contact = possible values : "OPEN", "CLOSED", "TIMEOUT"
- motion = possible values : "MOTION", "NO MOTION"
- illuminance = possible values : "BRIGHT", "DARK"
- buttoncount = cycles between values 1-15 , resets after 15
- entrancecount = cycles between values 1-3 , resets after 3
- outcount = cycles between values 1-3 , resets after 3
- lastmotion = NOT COMPLETED, will eventually be a timestamp
- lastcontact = NOT COMPLETED, will eventually be a timestamp

In order to test contact sensor
- leave the door open for at least 35 seconds, and see if HA updates contact properly
- leave the door closed for at least 35 seconds, and see if HA updates contact properly
- have no motion for 35 seconds, and see if HA updates motion properly
- have constant motion for 35 seconds, and see if HA updates motion properly
- push the button, wait 35 seconds, check if the buttoncount value changed (1 to 3)
- make it dark, wait 35 secs and see if HA updates illuminance properly
- make it bright, wait 35 secs and see if HA updates illuminance properly

In order to test motion sensor
- have no motion for 35 seconds, and see if HA updates motion properly
- have constant motion for 35 seconds, and see if HA updates motion properly
- make it dark, wait 35 secs and see if HA updates illuminance  properly
- make it bright, wait 35 secs and see if HA updates illuminance properly

BangerTech, 2021-09-06:
Hi and thanks for the work! 
I´ll try this out today and let you know.

Greets

Mike

devWaves, 2021-09-06:
I updated the branch to v6Alpha2 and I might actually have it working with contact and motion sensor. They are not longer doing the same things as meters, the ESP32 should update based on motion

the code now scans all the time and will send MQTT on updates from the contact sensors and motion sensors.

The only thing left it to collect the last motion timestamp and last contact timestamp. So then if a message is missed I can send one based on the last timestamp

devWaves, 2021-09-06:
I plan on sending a button push MQTT message whenever the button count changes

So you can use the button on the contact sensor to trigger other things in HA

BangerTech, 2021-09-06:
awesome that sounds perfect!! i was busy today but i´ll test it out tomorow morning. Thanks a lot!!

BangerTech, 2021-09-07:
so i´ve tested the contact sensor but i don´t get any data in mqtt.fx. I´ll try and flash the esp again maybe there was a error

BangerTech, 2021-09-07:
i´ve flashed the esp again now i´ve put in the motion sensor aswell but i get no informations about the contact and motion sensor, meter and bot works fine

BangerTech, 2021-09-07:
[image]

devWaves, 2021-09-07:
did you perform a reboot? for some reason I have to reboot sometimes to see new devices show up

if the MAC were at least detected the HA MQTT diwcovery should have happened

It works for me when I fake a bot as a motion/contact sensor, but it can detect bot

And they are all setup/configured in the switchbot app already?

devWaves, 2021-09-07:
The way that I tested was I faked a bot as a motion/contact sensor then I used the switchbot app to control the bot

when i used the switchbot app to turn the bot ON, it would update in HA to ON after about a second

I assumed this would simulate a Motion=ON from a contact/motion sensor

BangerTech, 2021-09-07:
yes the contact and the motion sensor are setup in the app. wrote the mac adresses in the code the same way like the meter and bot so it should be fine. just did a reboot on the esp and checked everything again. 
so if i scan in mqtt.fx i don´t the a channel for example switchbot/esp32/motion/motionone..... so i´ve tried to subscribe to one manually but no data is coming. i assume the state channel should be the same way right the one from the meter or the bot right? like switchbot/esp32/motion/motionone/state

devWaves, 2021-09-07:
Hmm...

I don't have state for motion or contact. I can add them though. The contact has both motion and contact, so there are technically 2 states

The first MQTT message you would see when the device is discovered would be on your MQTT discovery topic. Example:
```
homeassistant/
homeassistant/sensor/contactone/
homeassistant/sensor/contactone/
homeassistant/binary_sensor/contactone/
homeassistant/binary_sensor/contactone/
homeassistant/sensor/contactone/contact/config
homeassistant/sensor/contactone/buttoncount/config
```

the issue seems like the MACs are not being discovered if you aren't at least getting those

Then once after that the data would all be on these topics
```
switchbot/esp32/contact/contactone/
switchbot/esp32/motion/motionone/
```

BangerTech, 2021-09-07:
hmm yeah the mac´s are not discovered. 
i don´t understand why it should be the same way like the bots, curtains & meters are discovered right?

devWaves, 2021-09-07:
ya I can only read the available documentation at this point. It sucks switchbot doesn't sell in Canada and I have to pay extra in fees extra to order them

I made the code so that if there are contact/motion sensors it will scan all the time, if there are none in the lists it will work the same way as before and only scan when needed. Even when scanning all the time though, it only sends MQTT updates when needed

back to the drawing board I guess....

If you turn on printSerialOutputForDebugging, you can watch the Arduino IDE Serial monitor logs to see if you see you MAC addresses in the lists as they fly by
```
static bool printSerialOutputForDebugging = true;  
```

you can unplug the ESP32 to stop the logging, then copy/paste the contents to notepad and search the text for your MACs

devWaves, 2021-09-07:
crap... I think I know what I did wrong. They dont work the same as bots. You actually have to connect and subscribe to BLE notifications. Hopefully I can make that work

oh well the other changes I made are still a good addition

BangerTech, 2021-09-07:
[image]

BangerTech, 2021-09-07:
so the mac´s are both in there. ......b8:3d is the contact sensor and ....21:16 is the motion sensor.

ahh ok so the ble notification would be the solution? 
yaa i hope so too! but i think you can ;-)

devWaves, 2021-09-07:
so ya they should still work the way I did it, but wont be instant status. Instant status can only be done by subscribing to BLE notifications. Ill see if that works. One device would work, but not sure if subscribing to multiple devices works

Maybe the BLE scan timeout is too short to get all the data because from your screenshot it looks like that is the issue. They are being found by MAC but not matching to the service uuid

devWaves, 2021-09-07:
You can try changing these values. Specifically setWindow

```
pScan->setInterval(45);
pScan->setWindow(1000);
```

I believe this will make it scan longer, and might get more data from the device

BangerTech, 2021-09-07:
so setWindow to 1000 didn´t help.
no i don´t think contact or motion supports password i don´t see any of that option in the app

devWaves, 2021-09-07:
ok. not sure why. According to documentation that should work

I will make some changes to force add the devices if the MACs match, then I will add the BLE connect/notify stuff

from your screenshot, the data packet only has one byte = d

Can you search your logs again to see if there are more entries for your MACs. The lines under the MAC address will say UUID: .... data: ......

I am hoping to see more on the data part

maybe you can try a passive scan instead of active scan

set this to false. I have it as true
```
pScan->setActiveScan(false);
```

BangerTech, 2021-09-07:
[image]
[image]

BangerTech, 2021-09-07:
so i did change the ...ActiveScan to false. 
hope this screenshot i useful. the adress: .....:b8:3d is the contact sensor

devWaves, 2021-09-07:
No more usueful info in the screenshots. thanks tho. I still have an idea though, just need to code it up

devWaves, 2021-09-07:
Can you try using a GATT tool to get a screenshot of the service data?

the tool I just tried for Android is called "nRF Connect for Mobile". you can use any app that will provide the service data. Within the app, under scanner, start a scan and find your device. Click on it to expand it and take a screenshot

Do you have cloud access enabled? maybe you need to do that for it to broadcast? not sure why it isnt seeing all the data in the scan

devWaves, 2021-09-07:
if you have your switchbot hub connected to them, turn your switchbot hub off. They require a constant connection so maybe the switchbot hub being connected to them messes things up

I know that only one BLE connection can be made to a bot at a time, so that would make sense

BangerTech, 2021-09-07:
soo the hub is not installed so that shouldn’t be a problem. 

I’ll enable cloud function to see whats up ;-)

BangerTech, 2021-09-08:
[image]

devWaves, 2021-09-08:
ok that looks correct but without the service UUID. I guess it doesnt broadcast while not connected to it?

the data part looks correct so I can use that. I'll make some changes later today to add the devices if the MAC is found

also your RSSi connections are not good at least with your phone. anything below -80 is not good. 
From experience with the bot, a connection requires a good rssi value

devWaves, 2021-09-08:
I cpmmited a new change that will ignore looking for the service UUID and add it if it find the MAC address

You should at least get the MQTT discovery device created now. The values is the question if its correct or not

Try it out and let me know

BangerTech, 2021-09-09:
you sir... are great ;-) so now i get every data correct! 
so for example the contact sensor it gets data ever second. 
get the rssi, batt, motion with (Motion/NO Motion), contact: OPEN/CLOSED, light: (BRIGHT/DARK), entrancecount: 1/2....., outcount: 0/1/2..., buttoncount: 0/1/2....
maybe you could change the counts to a YES/NO or something and the button aswell. 

for the motion: updates every second too. 
rssi, batt, motion: MOTION/NO MOTION, led: ON/OFF, sensedistance: SHORT/MEDIUM/LONG, light: BRIGHT/DARK

BangerTech, 2021-09-09:
[image]
[image]

BangerTech, 2021-09-09:
the only wrong function i´ve discovered is when i´m controlling the switchbot bot via the app i get no mqtt message on switchbot/esp32/bot/switchbotone/attributes if i send a mqtt message switchbot/esp32/bot/switchbotone/set ON then the attributes are updated to but the state is still OFF

BangerTech, 2021-09-09:
[image]

devWaves, 2021-09-09:
awesome! so I guess I do have some idea of what I am doing, lol

Ya I don't have the bot updating attributes when you call with switchbot app atm. I did that during testing though and it worked, so I will add it in the final release. The code only scans all the time if you have a contact sensor or motion sensor in the list. I will add a true/false value for people that want to scan all the time in order to get these kinds of updates

the outCount,entranceCount,buttonCount are all counts they are not booleans. Is it the number of times ......, then the value resets to 0 to start again. Thats how switchbot did it. I will add a binary sensor for when the value changes though

2 main things left to do
1. see if I can connect and subscribe to BLE notifications to get instant status, and leave it subscribed
2. figure out how to get last contact, last motion timestamps

BangerTech, 2021-09-10:
Hey 
just wanted to check if you need any more infos from my side? ;-)

devWaves, 2021-09-10:
I think I've got everything for now. I'll try to look at the other stuff this weekend

just let me know if you are seeing mqtt updates every second even if nothing changes, because you shouldnt. you will get updates on a rescan also

devWaves, 2021-09-12:
I will need you to test something and report back to me please

test this code with ONLY one contact sensor or one motion sensor (no other devices)
https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/v6MotionContactTesting/SwitchBot-BLE2MQTT-ESP32.ino

I need you to let me know if you see any messages on these topics (or whatever you used for topics names that match this)

**Topic 1)**
```
switchbot/esp32/contact/contactone/subscribetest
switchbot/esp32/contact/motionone/subscribetest
```

I am looking for at least one message with payload ``` "subscribed" ```

**Topic 2)**

```
switchbot/esp32/contact/contactone/gotsubscribedmsg
switchbot/esp32/contact/motionone/gotsubscribedmsg
```

payloads
```
"GOT MESSAGE"
"PROCESSED MESSAGE"
```

If you aren't getting any messages on the 2nd topic specifically, then it doesn't work the way I think it does

This code is only for testing and won't work fully correctly

BangerTech, 2021-09-12:
[image]
[image]

BangerTech, 2021-09-12:
so for the contact sensor i got 3 topics. 
- switchbot/esp32/contact/contactone/msglengthScan
- switchbot/esp32/contact/contactone/status
- switchbot/esp32/contact/contactone/subscribetest

see the payloads on the pictures, first one is when i set the payload decoded by plain text, second one is when i set payload decoded by json pretty format

BangerTech, 2021-09-12:
[image]

BangerTech, 2021-09-12:
got no ...../gotsubscribedmsg on a movement or by a physical change of the device

devWaves, 2021-09-12:
ok thanks a lot. It looks like my original way of doing it is the "proper" way, I just need to make some changes.

I assumed if you kept a connection active and subscribed to BLE notifications it would provide instant status update. But thats not the case from what I can tell

When using the app is it instant status update? is it just as fast with the way I coded it before? (with the esp32 similar distance as phone)

it looks like I was overthinking it

BangerTech, 2021-09-12:
no worries i do what i can ;-)
ya on the app it takes like a second so pretty instant ;-) kinda the same way like with the esp.

BangerTech, 2021-09-18:
hey buddy what´s up ;-)
did you got any chance to check the code yet? can´t wait to test it...

devWaves, 2021-09-18:
I actually just wrote up a new version. You can try it out
https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/v6MotionContact/SwitchBot-BLE2MQTT-ESP32.ino

It should have all the data elements for contact and sensor except for lastMotion and lastContact because I do not understand the switchbot documentation for lastMotion and lastContact.

I am trying to figure out an issue though. The ESP32 may randomly crash while scanning. If you notice a message like this....
```
switchbot/esp32
"{"status":"boot"}"
```

it means the ESP32 rebooted. If you get that and you didn't purposely restart the esp32, then that is the issue I am trying to figure out

I am trying to figure out if the constant scanning and having the OTA webpage up at the same time is causing the issue

devWaves, 2021-09-19:
I think I fixed all the issues. try v6Alpha5 from https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/v6MotionContact/SwitchBot-BLE2MQTT-ESP32.ino and let me know please

you will want to delete your previous contact and motion devices if you created any

If you don't have issues with this version I will release on main branch

I will add some changes after for getting the last trigger contact/motion timestamps but I don't understand the documentation at the moment and I don't think anyone else has coded for that yet either to reference. It is a //TODO in the python code

1) the contact/motion sensors will update whenever there are changes
2) the bot/curtain/meter will only scan for changes when requested. So if you use the switchbot app you need to wait for the next scan to get the latest status. You can set rescanTime to a low value and it will scan somewhat continuously if you want quicker updates if you are also using the switchbot app

just keep an eye out for a error reboot mqtt message. Those are not good
```
switchbot/esp32
"{"status":"boot"}"
```

BangerTech, 2021-09-19:
thanks!
sooo for me i get no reboots, runs for like 3 hours now and no reboot so far. 
i´ve tested each device and the sensor work fine, i get all informations and they change values immediately. But i get into some changes which you could make later. 
the only device which makes problems right now is the bot. 
so for instance if control the bot via mqtt with publishing messages: (in mode:press)
if i send ON i get esp32 : status: idle then immediately esp32: status: scanning and a state change in the attributes to ON . 
But if i send OFF i get  esp32: status : idle then immediately esp32: status: scanning and a attributes message but without a change of state and the bot doesn´t do anything. If i send a PRESS message the bot works all the time but no state change in the attributes i get a attributes message though. 

ahhh just saw that you updated the code so i do some test again ;-) so don´t take this upper stuff ;-)
i´ll be testing the new one

BangerTech, 2021-09-19:
soo now for the bot: (press mode)
- with mqtt publish PRESS bot works all the time i get a message : set(PRESS) - state(OFF) - set (PRESS) - state (OFF) - status (status:connected, command: PRESS) - status (status:commandSent, command:PRESS) - state (OFF) - attributes (rssi, mode: PRESS, state: OFF, batt) - state(OFF)
- with mqtt publish ON i get message - set (ON) - state (OFF) - status ( status: connected, command: ON) - status (status: commandSent, command: ON) - assumedstat: (ON) - status (status: success, value: 5, command:ON) - state (OFF) - attributes: (rssi: -66, mode: Press, state: OFF, batt: 95) - state: (OFF)
all message in 10 secs. 
- with mqtt publish OFF the bot doesn´t do anything and i get message - set(OFF) - state (OFF) - attributes (rssi, mode:Press, state:OFF, batt) - state (OFF) so a few message less and the bot doesn´t work. 

soo then i change to switch mode: 
- with mqtt publish ON bot works and  in the app it goes to ON aswell messages: - set(ON) - state(OFF) - status(status:connected, command: ON) - status(status: commandSent, command: ON) - assumedstate(ON) - status(success, value:1, command:ON) - state(OFF) - assumedstate(ON) - attributes(rssi, mode:Switch, state:ON, batt) - state(ON)
- with mqtt publish OFF bot works and in the app it goes to OFF, messages: set(OFF) - state(OFF) - status(status:connected, command:OFF) - status (status: commandSent, command: OFF) - assumedstate(OFF) - status(status:success, value:1, command:OFF) -state (OFF) -  assumedstate(OFF) - attributes(rssi, mode:Switch, state:OFF, batt) - state(OFF)

devWaves, 2021-09-19:
yes that all sounds correct. an "OFF" command should be ignored if the bot is in press mode

when the bot is in press mode, only send "PRESS" or "ON"

the state will return back to "OFF" when in press mode.

devWaves, 2021-09-19:
you should see a message under the button topic for contact devices. you can use the button on the contact device to trigger other things in your smarthome (non switchbot) whenever the payload is "PUSHED"

BangerTech, 2021-09-19:
if i control the bot via Switchbot APP: 
PressMode: 
- bot is off and i press the button in the app (changed rescantime to 240) get the message in within 240secs: - attributes(rssi, mode: PRESS, state: OFF, batt) - state(OFF) 
so no change in the state. 
- bot (controlling device is on) and i press in the app again to turn it off i get - attributes(rssi, mode:Press, state:OFF, batt) - state(OFF)

ok just got your message, ok got it would be cool if it jumps between on and off even in pressmode to get some kind of knowing what state the device you want to control is in or am i wrong? maybe its not possible

devWaves, 2021-09-19:
yep you can do that. This is when you want to add your device also to the simulated on/off list

this is already a feature

BangerTech, 2021-09-19:
ahh sorry just saw it in the code ;-)

devWaves, 2021-09-19:
when the device is also in the botsSimulateONOFFinPRESSmode list you can use "OFF" command

[9 more replies]

closed 2021-09-19 by devWaves