devWaves/SwitchBot-MQTT-BLE-ESP32 #29: Flashed to ESP-WROOM-32 and keeps rebooting

abus3ood, 2021-09-23:
Flashed on ESP32 Dev Module (ESP-WROOM-32) and it Keeps rebooting.

it loops over status 'boot','scanning', I don't know where to start :/ .

10:22:42.975 -> ets Jun  8 2016 00:22:57
10:22:42.975 -> 
10:22:42.975 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
10:22:42.975 -> configsip: 0, SPIWP:0xee
10:22:42.975 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
10:22:42.975 -> mode:DIO, clock div:1
10:22:42.975 -> load:0x3fff0030,len:1240
10:22:42.975 -> load:0x40078000,len:13012
10:22:42.975 -> load:0x40080400,len:3648
10:22:42.975 -> entry 0x400805f8
10:22:45.584 -> E (4696) wifi:Association refused temporarily, comeback time 1024 mSec
10:22:57.257 -> Switchbot ESP32 starting...
10:22:57.257 -> Set printSerialOutputForDebugging = true to see more Serial output
10:23:01.638 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
10:23:01.638 -> 
10:23:01.638 -> Core  1 register dump:
10:23:01.638 -> PC      : 0x40094e8d  PS      : 0x00060e33  A0      : 0x40092b23  A1      : 0x3ffd5690  
10:23:01.638 -> A2      : 0x3ffc5384  A3      : 0x3ffbf278  A4      : 0xb33fffff  A5      : 0x00000001  
10:23:01.672 -> A6      : 0x3ffc52fc  A7      : 0x3ffc5190  A8      : 0x53003ffc  A9      : 0x00000000  
10:23:01.672 -> A10     : 0x3ffc5300  A11     : 0x00000012  A12     : 0x53140000  A13     : 0x00000048  
10:23:01.672 -> A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
10:23:01.672 -> EXCVADDR: 0x00000048  LBEG    : 0x40096819  LEND    : 0x40096828  LCOUNT  : 0x00000001  
10:23:01.705 -> 
10:23:01.705 -> 
10:23:01.705 -> Backtrace:0x40094e8a:0x3ffd56900x40092b20:0x3ffd56d0 0x40092bb1:0x3ffd56f0 0x400e7a4a:0x3ffd5710 0x400f19c9:0x3ffd5770 0x400f263f:0x3ffd5940 0x400f2f2f:0x3ffd5a40 0x400f339f:0x3ffd5bd0 0x400fbeae:0x3ffd5c50 0x40121e48:0x3ffd5cb0 
10:23:01.705 -> 
10:23:01.705 -> 
10:23:01.705 -> 
10:23:01.705 -> 
10:23:01.705 -> ELF file SHA256: 0000000000000000
10:23:01.705 -> 
10:23:01.705 -> Rebooting...

abus3ood, 2021-09-23:
these are all the warnings:

Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:659:69: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayPos[] =  {0x57, 0x0F, 0x45, 0x01, 0x05, 0xFF, NULL};
                                                                     ^
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:661:56: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayHoldSecs[] = {0x57, 0x0F, 0x08, NULL };
                                                        ^
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:662:60: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayBotMode[] = {0x57, 0x03, 0x64, NULL, NULL};
                                                            ^
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:662:60: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:664:68: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayPressPass[] = {0x57, 0x11, NULL, NULL, NULL, NULL};
                                                                    ^
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:664:68: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:664:68: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:664:68: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:665:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayOnPass[] = {0x57, 0x11, NULL , NULL, NULL, NULL, 0x01};
                                                                        ^
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:665:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:665:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:665:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:666:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayOffPass[] = {0x57, 0x11, NULL, NULL, NULL, NULL, 0x02};
                                                                        ^
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:666:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:666:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:666:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:667:74: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayGetSettingsPass[] = {0x57, 0x12, NULL, NULL, NULL, NULL};
                                                                          ^
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:667:74: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:667:74: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:667:74: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:668:84: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
[... cut ...]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:670:82: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino: In function 'bool writeSettings(NimBLEAdvertisedDevice*)':
/Users/ap/Downloads/SwitchBot-MQTT-BLE-ESP32-main/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:1295:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^

devWaves, 2021-09-23:
first, what kind of devices are you trying to use? I can't diagnose those issues without seeing your code. definitely something with what you added/changed or something with your ide

you can send me your code on HA forum or reddit or something if you want

abus3ood, 2021-09-23:
I have just downloaded your code and verified it in Arduino IDE and it gave me all these warnings.
I am using the ESP32 dev module (board in Arduino IDE)

devWaves, 2021-09-23:
there are no issues with the code. It is something with your setup. Did you follow the instructions of settings up the arduino ide for your esp32 board type? Did you modify the code ? because u need to

are you using a fresh arduino ide install or have you used arduino ide before? sometimes you can  get get conflicting libraries. if u have use arduino ide before, Delete the Documents -> Arduino -> Libraries to remove all your libraries. Delete the User -> AppData -> Local -> Arduino15 folder to remove all the preferences. Uninstall Arduino IDE. Install a fresh one. Then try again

devWaves, 2021-09-25:
I am closing this issue since it isn't an issue with the code, but feel free to continue to ask questions here

closed 2021-09-25 by devWaves

abus3ood, 2021-09-25:
Sure, I downloaded your code again and tried to be careful but warnings
still show in the Arduino IDE but at least the switchbot pusher shows in HA
and reports almost all attributes but, when pushing a button to affect the
switch nothing happens.

Thanks anyway for you code and effort to maintaining this code and
responding to issues.

devWaves, 2021-09-25:
what is the rssi linkquality value you get?
if in the 80s-90s you'll want to move the esp32 closer to the bot

so you only have one bot ?
did you make sure that only the bot mac address is in the list? any other ones should be commented out or removed

what are you seeing as a device and entities in HA? screenshot please

Did you try with a fresh arduino install? you should get zero warnings when compiling/uploading

what are you using for device names etc? 

if you send me the code you used I can probably tell you what the issue is pretty quickly. Just replace the passwords etc. you will need to make sure the formatting is kept in what you send

abus3ood, 2021-09-26:
I have the bot next to the esp32 -maybe 30cm away- so I don't think its a weak signal issue.
I only have one bot and I have removed the rest from the code.

I see in HA this:
https://ibb.co/tXQkBs2
https://ibb.co/2WjKJY4

HA -Core- and MQTT are running in Docker on my NAS.
 
it might be my setup If you see nothing wrong in the code below I will redo my HA setup on a full RPi4 and see if anything works after that. 
`
#include <NimBLEDevice.h>
#include <EspMQTTClient.h>
#include <ArduinoJson.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <Update.h>
#include <CRC32.h>
#include <ArduinoQueue.h>

/****************** CONFIGURATIONS TO CHANGE *******************/

/********** REQUIRED SETTINGS TO CHANGE **********/

/* Wifi Settings */
static const char* host = "esp32";                          //  Unique name for ESP32. The name detected by your router and MQTT. If you are using more then 1 ESPs to control different switchbots be sure to use unique hostnames. Host is the MQTT Client name and is used in MQTT topics
static const char* ssid = "***";                           //  WIFI SSID
static const char* password = "***";                   //  WIFI Password

/* MQTT Settings */
/* MQTT Client name is set to WIFI host from Wifi Settings*/
static const char* mqtt_host = "192.168.100.16";                       //  MQTT Broker server ip
static const char* mqtt_user = "";                         //  MQTT Broker username. If empty or NULL, no authentication will be used
static const char* mqtt_pass = "";                         //  MQTT Broker password
static const int mqtt_port = 1883;                                  //  MQTT Port
static const std::string mqtt_main_topic = "switchbot";             //  MQTT main topic

/* Switchbot Bot Settings */
static std::map<std::string, std::string> allBots = {
   {"keybot", "E9:63:08:29:19:B5"}
   
};

/* Switchbot Curtain Settings */
static const int curtainClosedPosition = 10;    // When 2 curtains are controlled (left -> right and right -> left) it's possible one of the curtains pushes one of the switchbots more open. Change this value to set a position where a curtain is still considered closed
static std::map<std::string, std::string> allCurtains = {
  /*{ "curtainone", "xX:xX:xX:xX:xX:xX" },
    { "curtaintwo", "yY:yY:yY:yY:yY:yY" }*/
};

/* Switchbot Meter Settings */
static std::map<std::string, std::string> allMeters = {
  /*{ "meterone", "xX:xX:xX:xX:xX:xX" },
    { "metertwo", "yY:yY:yY:yY:yY:yY" }*/
};

/* Switchbot Contact Sensor Settings */
static std::map<std::string, std::string> allContactSensors = {
  /*{ "contactone", "xX:xX:xX:xX:xX:xX" },
    { "contacttwo", "yY:yY:yY:yY:yY:yY" }*/
};

/* Switchbot Motion Sensor Settings */
static std::map<std::string, std::string> allMotionSensors = {
  /*{ "motionone", "xX:xX:xX:xX:xX:xX" },
    { "motiontwo", "yY:yY:yY:yY:yY:yY" }*/
};

[... 4224 lines cut ...]
        statDoc["status"] = "lowbattery";
        lastCommandWasBusy = false;
      }
      else {
        statDoc["status"] = "failed";
        lastCommandWasBusy = false;
      }
      statDoc["value"] = byte1;
      statDoc["command"] = aCommand;
      serializeJson(statDoc, aBuffer);
      client.publish(deviceStatusTopic.c_str(), aBuffer);
    }
  }

  NimBLEClient* pClient = nullptr;
  if (NimBLEDevice::getClientListSize()) {
    pClient = NimBLEDevice::getClientByPeerAddress(deviceMac);
    if (pClient) {
      if (pClient->isConnected()) {
        unsubscribeToNotify(pClient);
        //pClient->disconnect();
      }
    }
  }
}`

devWaves, 2021-09-26:
I can take a closer look when I am at my PC about warnings, but I just did a quick look through and noticed your bot password is commented out

you won't be able to control the bot if it has a password and you aren't using the correct one

this makes sense because you also dont get values for firmware, timers, holdsecs. They values require an initial connection to the bot (unlike the others). So a wrong password would do that also

also u have v6.1. get v6.2

devWaves, 2021-09-26:
ya I just compiled and uploaded the code you sent. I am getting no errors or warning. But as mentioned your bot password is commented out

If you are still getting errors or warning from the arduino IDE you should either try from another computer or completely wipe the arduino IDE stuff and try again. You need to delete libraries, not just uninstall/reinstall

From what you see in HA though, it looks like it compiled because you see the proper devices and entities

You don't need to redo HA, your HA is good

also you have an RSSI of -49 (which is REALLY good). The bot should action basically instantly with an RSSI like that

devWaves, 2021-09-26:
also you don't "need" your keybot in this list....

```
static std::map<std::string, int> botScanTime = {     // X seconds after a successful control command ESP32 will perform a requestInfo on the bot. If a "hold time" is set on the bot include that value + 5to10 secs. Hold time is auto added to default value. Default is 10+Hold sec if not in list
  { "keybot", 10 }
};
```

infact, since you are using simulate ON/OFF with an ON hold time of ```{ "keybot", 15 }```  the botScanTime value you put of ```{ "keybot", 10 }``` is not big enough.  This would mean the bot will hold for 15 seconds when OFF is pressed, but then after 10 secs and during that 15 seconds while holding the esp32 will try to scan for the latest status.... but this won't work because the bot is still holding for that 15 secs.

abus3ood, 2021-09-26:
Thanks for the instant reply, I did a find/replace all to replace
'switchbotone' with 'keybot' 😅 thats why you see the change everywhere .

I will pull your latest code and try a clean install of the IDE and see how
it goes from there and will keep you posted on the results.

Thanks and keep it up!