devWaves/SwitchBot-MQTT-BLE-ESP32 #30: missing sdkconfig.h

thematrixdev, 2021-09-24:
Trying to upload to Arduino it fails with the following error:

> In file included from /home/user/SwitchBot/SwitchBot.ino:320:
/home/user/Arduino/libraries/NimBLE-Arduino/src/NimBLEDevice.h:17:10: fatal error: sdkconfig.h: No such file or directory
   17 | #include "sdkconfig.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

I have found this page, but the sdkconfig.h is 404
https://forum.arduino.cc/t/fatal-error-sdkconfig-h-no-such-file-or-directory/562561

May I know how to fix the problem?
Thanks.

devWaves, 2021-09-24:
are you using a fresh arduino ide install or have you used arduino ide before? sometimes you can get get conflicting libraries. if u have use arduino ide before, Delete the Documents -> Arduino -> Libraries to remove all your libraries. Delete the User -> AppData -> Local -> Arduino15 folder to remove all the preferences. Uninstall Arduino IDE. Install a fresh one. Then try again

Based on the error you are trying to compile for a ESP-12E. You need to be using an ESP32. If you did step #2 from the instructions you will select the board type you are using from here....

[image]

I am using a Wemos D1 mini ESP32

devWaves, 2021-09-24:
https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/issues/15

same issue

thematrixdev, 2021-09-24:
Sorry I am not very familiar with the hardwares. Mine should be ESP8266. Should I use ESP32 instead?

> are you using a fresh arduino ide install or have you used arduino ide before? sometimes you can get get conflicting libraries. if u have use arduino ide before, Delete the Documents -> Arduino -> Libraries to remove all your libraries. Delete the User -> AppData -> Local -> Arduino15 folder to remove all the preferences. Uninstall Arduino IDE. Install a fresh one. Then try again
> 
> Based on the error you are trying to compile for a ESP-12E. You need to be using an ESP32. If you did step 2) from the instructions you will select the board type you are using from here....
> 
> [image]
> 
> I am using a Wemos D1 mini ESP32

closed 2021-09-25 by devWaves