devWaves/SwitchBot-MQTT-BLE-ESP32 #31: HomeAssistant warning "Payload is not supported"

stast1, 2021-09-29:
Log
```
2021-09-29 21:36:51 WARNING (MainThread) [homeassistant.components.mqtt.cover] Payload is not supported (e.g. open, closed, opening, closing, stopped): PAUSE
```
Please add stat_stopped: PAUSE to config topic to remove this warning.
```
client.publish((home_assistant_mqtt_prefix + "/cover/" + deviceName + "/config").c_str(), ("{\"~\":\"" + (curtainTopic + deviceName) + "\", " +
                 ...
                 + "\"stat_stopped\": \"PAUSE\", " +
                 ...
```
Thanks!

devWaves, 2021-09-29:
okay that makes sense. I don't have curtains. You're the first to mention it. Thanks for the input

I will release a v6.3 later today

JoepWi, 2021-09-29:
Thanks. I just didn't had the time  yet to post this issue here!

devWaves, 2021-09-30:
all good now. thanks for noticing the issue. fixed in v6.3

closed 2021-09-30 by devWaves