devWaves/SwitchBot-MQTT-BLE-ESP32 #34: Is it possible to support "remote" please?

thematrixdev, 2021-10-09:
This one. Thanks.
https://www.switch-bot.com/products/switchbot-remote

devWaves, 2021-10-09:
I was thinking about it and I might be able to reverse engineer if you send me a few screenshots from your android or iphone

I would need you to install an app that shows the advertised data published by the remote device MAC address. I would need a number of screenshots to see what changes in the data

The android app that I use is called "nRF Connect for Mobile" made by "Nordic Semiconductor ASA". There are many BLE apps that can provide this data though

If it publishes similar to the motion/contact sensor I can probably get it to work. The data should be fairly simple as we only care about 3 things

1) Battery
      Take at least 1 screenshot without pressing the remote. Does it publish advertised data?
2) a push up
       Push the up button a couple times and try to take a screenshot right after each press. At least 3 screenshots
3) a push down
      Push the down button a couple times and try to take a screenshot right after each press. At least 3 screenshots

Make sure to clearly define which screenshot is which when you send them to me.

I just need screenshots similar to this.... (this is for the bot, I would need them for remote)

[image]

devWaves, 2021-10-09:
I'm assuming that the button works by connecting directly to the device it is linked to.

I am hoping that it also advertises a count of the number of button pushes also. If it does not then I don't think I can support it. Otherwise I would have to fake the esp32 as a switchbot device and that wouldnt work or would be a ton of work

thematrixdev, 2021-10-10:
Battery level is 100%.
The service data is different today, strange.
[image]

Pressing the physical button on the switch-remote, clicking "refresh" in the app, service-data has not changed.

devWaves, 2021-10-10:
so in the service data the "64" is the battery. 64 hexidecimal converted to decimal value = 100 %

So there is no difference even between scans of pressing up and down? if so then I don't think I can support it

only thing I see different is the 42 changed to a 62. if you only pushed "up" the 4 in 42 could have changed to a 6, maybe meaning you pushed up 2 times?? but if you say it isnt changing all the time then probably not. 

it looks like I can only collect the battery level, which doesn't help much

it sounds like these are made to directly interact with another switchbot curtain or bot only. that sucks

My guess is you would connect to the remote with the switchbot app. It will program the remote with the bot MAC address you want to link it to and store that info in the remote. Then when you press the remote it will make similar BLE calls to control the bot

since it would be a device to device connection I can't capture that data

this is probably why there is no documentation for it

Only last thing I can think of is maybe if you don't already have it linked to a device it might broadcast different service data, but I doubt it

let me know if you ever see anything different though if you continue trying

thematrixdev, 2021-10-10:
I think you guess it correct. Thanks.

[image]

[image]

devWaves, 2021-10-10:
ok. thanks for the screenshots.

I might add it anyways at some point so that battery can be monitored