devWaves/SwitchBot-MQTT-BLE-ESP32 #37: Alexa support?

fabiosoft, 2021-11-11:
Do you plan to add Alexa/Google home support?

Maybe with [FauxmoESP](https://github.com/vintlabs/fauxmoESP) is easy, even for multiple switchbots

devWaves, 2021-11-11:
I wasn't planning of having it as part of the main branch, but I added a separate branch for Alexa if you want to test it out. I just quickly wrote this and would need someone to test since I don't own alexa/echos. I have google homes which don't have a similar integration

https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/alexaBranch/Arduino%20IDE%20Files/SwitchBot-BLE2MQTT-ESP32.ino

You will need to add the FauxmoESP library and also manually add the AsyncTCP library using the zip https://github.com/me-no-dev/AsyncTCP/archive/master.zip

The code currently still requires that you have MQTT setup and working, but you should be able to make the calls directly from Alexa for bots and curtain (ON,OFF). Position isn't setup to work through alexa yet

The standard code heavily expects MQTT to be connected, so I would need to remove those checks in order to not use MQTT

Again, I coded this with zero Alexa testing since I don't own one, so good luck

fabiosoft, 2021-11-12:
Hi, at home i own an Alexa Echo dot so i will test soon...according docs should be easy to implement percent position.
Maybe i will try to issue some voice commands to alexa and try to write some code.

I'll let you know :)

devWaves, 2021-11-12:
I can do the percentage part so u dont need to worry about that, the code now is just a preliminary test to see that it works for on/off. I might switch to espalexa... not sure yet

If it works I"ll make changes so you can disable the need for mqtt

fabiosoft, 2021-11-12:
okok, anyway i'll use both alexa (voice) and mqtt with my hub.
If you need some logs and prints about input parameters provided by alexa...let me know.

Did you find some improvements to use espalexa over Fauxmoesp? Maybe light bulb colors?

devWaves, 2021-11-12:
Without an alexa/echo to test with it is difficult to tell if one is better than the other (fauxmoesp or espalexa). I know aircoookies WLED work and espalexa is slightly easier to install (no zip to download). espalexa didnt look like it was setup for dynamic device names though

fabiosoft, 2021-11-13:
According to my experiments they are pretty mush the same, maybe the espalexa is more oriented to colors and brightness...as well as percentage but fauxmoesp also support boolean state and percent but colors are in a separate branch. The developer says he is not very happy with implementation.

Anyways both solutions seem to work good

devWaves, 2021-11-27:
@fabiosoft have u have the chance to test? if it works I'll look at removing the mqtt requirement

devWaves, 2021-12-10:
Closing this for now. Can be reopened later if needed

closed 2021-12-10 by devWaves