devWaves/SwitchBot-MQTT-BLE-ESP32 #38: Switchbot remote support?

mtgpcs, 2021-11-18:
Hello,

Current using two Switchbots and it works great! Thank you. Wondering if you had seen the Switchbot [remote](https://www.switch-bot.com/products/switchbot-remote) ? I am thinking to pick one up on the sale and use it as a physical switch. Would this be easy to support?

devWaves, 2021-11-18:
already request here #34. I can only support battery for the remote

The remote links directly with another switchbot product so it can't be used to trigger non switchbot products. The data between remote and switchbot bot/curtain can't be intercepted

If you want to control other non-switchbot devices and the bot/curtain at the same time by a single button push you can do this with a zigbee, wifi or zwave button

closed 2021-11-18 by devWaves

mtgpcs, 2021-11-18:
Thanks for the quick reply. I didn't see the closed tab. I'll try that. I have Shelly Button 1 that are wifi that work well with the two I already use.

Again thanks for the project! Worka great! Way better than HA default integration.

devWaves, 2021-11-21:
if it makes sense near a door, another option is to use a switchbot door contact sensor. With the esp32 code you can use the button on the contact sensor to trigger non switchbot stuff

this is a good option if u are using switchbot curtains on a patio door and the esp32 code already