devWaves/SwitchBot-MQTT-BLE-ESP32 #39: Curtains wont move

gkyprios, 2021-11-20:
Hi guys, let me start by saying that this is my first ESP32 project. Apologies for any dumb questions.

I have 4 x curtainbots in the same room that I would like to control with my M5Stack Atom Lite. I followed all of the instructions on the Git page and also used KPeyanski's youtube video. I am using v6.4. I can see all 4 devices and their entities in HA. The values of all the sensors exactly match the Switchbot app. So it would appear that the M5 is on wifi & BLE is working.

However when I use HA to send an instruction to open, close or even just move the curtains to a % - nothing happens. If I watch the MQTT messages using MQTT Explorer, I can see that a SET topic is generated for the device and the POSITION changes to that value. However the POS in ATTRIBUTES never changes. The curtainbot doesnt move and after about 5-10 seconds the POSITION changes back to its original value.

Of course, I can make the bots move with the Switchbot App so there isnt anything wrong with the devices. Would you mind helping me please? Let me know what details you need and how I generate them please.

devWaves, 2021-11-20:
What is the rssi linkquality value for each curtain? shown in attributes

Have you tested with the ESP32 close the the curtains?

Does the LED light on the ESP32 turn off eventually after boot? The LED will stay on until it finds all devices. So make sure to remove or comment out MACs for devices that don't exist. The devices cannot be controlled until the initial scan to find all devices is complete. By default the LED will remain on all the time if you have motion/contact sensors, but will turn off if you only have bot/curtain/meter. The LED light means it is scanning or controlling

gkyprios, 2021-11-21:
Hi dW.

RSSI for each curtain below:

deck_south
attributes = {"rssi":-60,"calib":true,"batt":54,"pos":100,"state":"OPEN","light":3}
position = {"pos":100}
state = OPEN

deck_north
position = {"pos":100}
attributes = {"rssi":-62,"calib":true,"batt":64,"pos":100,"state":"OPEN","light":2}
state = OPEN

patio_north
position = {"pos":100}
attributes = {"rssi":-88,"calib":true,"batt":66,"pos":100,"state":"OPEN","light":1}
state = OPEN

patio_south
position = {"pos":100}
attributes = {"rssi":-66,"calib":true,"batt":55,"pos":100,"state":"OPEN","light":1}
state = OPEN

The ESP32 is less than 5m from each curtainbot with no obstacles. Is that close enough? Otherwise I would need an extension cord,etc.

The LED - this is a problem I had not noticed. The LED doesnt turn on at all during boot or operation. I had an error during the Verify process about the LED but this went away when I made the following changes:

//#define LED_PIN LED_BUILTIN 
#define LED_PIN 27      

Do I have this set right? The Git URL for my board:

https://github.com/m5stack/M5Atom

Thanks dW.

devWaves, 2021-11-21:
so I just checked and with the M5Atom the led wont work the way its coded because pin 27 is an addressable led, not a binary on/off led, but it shouldnt break stuff

do you see any of these mqtt messages when you send open?...

topic:
<ESPMQTTTopic>/curtain/<name>/status
		
                    Example payload:
                      - {"status":"connected", "command":"OPEN"}
                      - {"status":"errorConnect", "command":"OPEN"}
                      - {"status":"errorCommand", "command":"NOTVALID"}
                      - {"status":"commandSent", "command":"OPEN"}
                      - {"status":"busy", "value":3, "command":"OPEN"}
                      - {"status":"failed", "value":9, "command":"OPEN"}

gkyprios, 2021-11-21:
Noted about the LED. I wont bother with it.

For the messages, here is an example of an open curtain that I'm trying to close;

21/11/2021 12:33:51 PM 
OPEN
21/11/2021 12:33:43 PM(-8.42 seconds) 
CLOSE
21/11/2021 12:33:00 PM(-43 seconds) 
OPEN
21/11/2021 12:32:51 PM(-8.35 seconds) 
CLOSE

I send a command from HA to CLOSE the curtain but it doesnt move and a message comes back as OPEN.

[image]

Also from the topic for the M5Atom:

21/11/2021 12:33:52 PM 
{"status":"idle"}
21/11/2021 12:33:51 PM(-1.21 seconds) 
{"status":"scanning"}
21/11/2021 12:33:50 PM(-0.15 seconds) 
{"status":"boot"}
21/11/2021 12:33:02 PM(-48.89 seconds) 
{"status":"idle"}
21/11/2021 12:32:59 PM(-2.31 seconds) 
{"status":"scanning"}
21/11/2021 12:32:59 PM(-0.18 seconds) 
{"status":"boot"}

[image]

Is that booting after each attempt to close the curtain? Am I reading that right or this expected behaviour?

devWaves, 2021-11-21:
if you see boot often then something is crashing the esp32, maybe it is the LED code? trying to set high low on an rgb LED?

you can watch the serial output in arduino ide by setting printSerialOutputForDebugging = true, you might see something there

devWaves, 2021-11-27:
Did you find anything new? u can send me a message on HA forum or here of your exact code and I'll see if anything stands out

gkyprios, 2021-11-28:
Hey dW.

Sorry. I havent had time this week to work on it at all. I was thinking that I could just disable the LED as I dont need it. But when I commented out both lines I got an error when it was compiling.

I've attached my entire code now. Can you look it over please?

Thanks in advance.

[SwitchBot-BLE2MQTT-ESP32.zip](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/files/7613276/SwitchBot-BLE2MQTT-ESP32.zip)

devWaves, 2021-11-29:
The code you used looks correct. If you are using curtains that are paired together, make sure to only include the pairs primary MAC address in the allCurtains list

If HA is finding all the curtains, then your MAC address would be correct

Since the atom esp32 doesn't have an LED to use, maybe try by commenting out every line that references LED_PIN . There should be 19 lines to comment out

so try by commenting out lines like these....

// pinMode (LED_PIN, OUTPUT);
// digitalWrite(LED_PIN, ledONValue);
// digitalWrite(LED_PIN, ledOFFValue);
//#define LED_PIN LED_BUILTIN 
//#define LED_PIN 2

gkyprios, 2021-11-30:
Thanks dW,

I tried commenting all 19 lines but that didnt seem to work. I am getting some errors during the verify stage that I could send to you privately if you wanted to fix this specific issue that the M5 Atom seems to have.

In the meantime, my D1 Minis arrived this morning. I got my first one running for all 4 curtains in the Living Room today. So nice.

I also got a second one running for my paired curtains in the bedroom that part from the middle. Thats even better!

Thanks so much for your work on this project and for your help with my issues.

devWaves, 2021-11-30:
cool. Ya so something specific to the m5atom esp32 that doesn't like the code since you got it working with D1 mini

if you are on reddit or the HA forums you can send me the serial logs there. or add here if the info isn't sensitive

devWaves, 2021-12-10:
closing the issue for now. If you find anything specific to the atom m5 esp32 let me know. Cheers

closed 2021-12-10 by devWaves