devWaves/SwitchBot-MQTT-BLE-ESP32 #42: Support slider control mode single side

iMicknl, 2021-12-03:
Thanks for providing this awesome code! :-). 

I combined two Switchbot Curtains 2 and I would like to have the same options as the app. To control both sides via a single cover entity (in Home Assistant), but also to control both sides separately via cover entities. 

Currently I added both mac addresses and the behavior is like this via SwitchBot-MQTT-BLE-ESP32:

- Left: this will move both sides (as left is the main switchbot in the switchbot app)
- Right: this will move only the right side

It seems that there is no way to only move the left side anymore via HA/mqtt, while it is possible via the original app.

By the way, I will have a look if I can port parts of the Curtain code to ESPHome as well in the coming months. I just received the Switchbot Curtain Urail 2.

JoepWi, 2021-12-04:
I solved your question by configuring and calibrate both sides as separate curtains with the SwitchBot app.

iMicknl, 2021-12-04:
Thanks, I thought about that as well, however that would mean that I need to work with a (combined) cover template in Home Assistant. Did you already create that code?

The nice thing of them combined is, is that they will be back in sync if you move them together via the Switchbot app.

JoepWi, 2021-12-04:
I abondonned totally the SwitchBot app. I will only use it in te future when I need to re-calibrate.
I have 13 units steered by 2 ESP32s.
And I created:
- script to close and open all sequential.
- An automations reaction on sunset plus a fixed time offset calling the close script
- An automation calling the open script at a fixed time.
- Two buttons close all & open all.

And I have just a dash with all the curtains and the two buttons.

devWaves, 2021-12-05:
although this is probably possible, it doesn't look like the documentation is available for how to achieve this and what commands to use etc

when you pair 2 curtains together from the switchbot app, the calls to the main MAC address operate both curtains. There might be commands for specifically controlling individual curtains through the main mac but I do not know what those would be

it looks like to achieve this you would not link the curtains in the switchbot app and then create a 3rd device which is a combination of the 2

This could work, but wouldnt be as seamless as linking the 2 curtains since the esp32 would send 2 commands instead of one. This will probably cause a small delay between the 2 curtains moving in sync

iMicknl, 2021-12-05:
Thanks @devWaves! Are there ways for me to research / reverse engineer the behavior of the official Switchbot app? Happy to see if I can mimic this behavior.

devWaves, 2021-12-05:
@iMicknl I am only aware of how to reverse engineer the service data broadcasted by the switchbot devices. I am not aware of how to intercept or determine commands between the switchbot app/hub to switchbot devices. I would only be able to do this through trial and error by sending commands and seeing how the devices react which would be tedious

I will close the ticket for now, but if you find anything let me know

closed 2021-12-05 by devWaves