devWaves/SwitchBot-MQTT-BLE-ESP32 #46: Press Mode on SwitchBot Bot

jeffconrad18, 2021-12-19:
Hey DevWaves,
it would be very cool if you add the new Home Assistant Feature ["button entity"](https://www.home-assistant.io/blog/2021/12/11/release-202112/), so if the Press-Mode is on, it appears as a Button in HA!
Thanks for your work!
Jeff

devWaves, 2021-12-19:
I will think about it, but it is only a visual thing. The default state of the switch is OFF and it acts as a button already

The thing about the bot is you can dynamically change the mode setting, which would mean resending the HA MQTT discovery messages and deleting the old devices in order to see it as a button vs switch when this happens. Not that you would change the mode often once setup, but it confuses things a little

devWaves, 2022-01-16:
button entity option added to v6.6

depending on the bot type you pick, it will create the proper HA entity

```
/* Switchbot Bot Device Types - OPTIONAL */
/* Options include: "switch", "light", "button" */
static std::map<std::string, std::string> allBotTypes = {     // OPTIONAL - (DEFAULTS to "switch" if bot is not in list) - Will create HA entities for device types
 /* { "switchbotone", "switch" },
    { "switchbottwo", "light" },
    { "switchbotthree", "button" }*/
};
```

closed 2022-01-16 by devWaves