devWaves/SwitchBot-MQTT-BLE-ESP32 #48: SwitchBot Curtain Mode missing

garv3, 2022-01-02:
SwitchBot Curtain supports two motion modes: "Silent Mode" and "Performance Mode".
It would be great if these would be available (get and set) through MQTT. Is this possible?

devWaves, 2022-01-02:
I can do a 'get', but the 'set' part isn't documented here  https://github.com/OpenWonderLabs/python-host/wiki/Curtain-BLE-open-API

so until I see the documentation, or if you want to try and figure out the command using a tool like "nrf connect" there isn't much I can

I will reopen if someone can provide the command to set the "Action mode" performance mode vs silent mode

closed 2022-01-02 by devWaves

iMicknl, 2022-01-03:
@devWaves did you use nrf connect for other debugging purposes? I am happy to help with a tool like nrf connect, however I was wondering if there were any specific tricks you performed. Any documentation that will help?

devWaves, 2022-01-03:
hey @iMicknl and @garv3

nrf connect is a good tool for testing sending/reading BLE data without needing to write any code, but I don't have any special tricks

Most of the commands to send data to curtains are documented here https://github.com/OpenWonderLabs/python-host/wiki/Curtain-BLE-open-API but it seems some stuff is missing. The only way I know to figure out the rest is to take some guesses based on the existing commands documented. That could take some work depending on how different the commands are

Here is how to send the commands to a curtain MAC address with nrf connect...

This is the open command byte array ``` {0x57, 0x0F, 0x45, 0x01, 0x05, 0xFF, 0x00} ```    0x00 = 0
This is the close command byte array ``` {0x57, 0x0F, 0x45, 0x01, 0x05, 0xFF, 0x64} ```    0x64 = 100

1. Open nrf Connect app
2. Scan and find your curtain MAC address
2. Connect to it
3. 
[image]

4.
[image]

getting the current status of the curtain mode is documented and in the ESP32 code can be retrieved the same way I retrieve switchbot bot settings like hold time and firmware

```
0x46 Curtain get command
Action mode A: 0-performance mode, 1-silent mode, other-invalid value, the default invalid value is 1 for all bit values.
```

conrad784, 2022-09-04:
> I will reopen if someone can provide the command to set the "Action mode" performance mode vs silent mode

I did a Bluetooth snoop and got back those values:
# set performance mode
57 0f 45 04 01 01 00 00
# set silent mode
57 0f 45 04 01 01 01 01

When refreshing the settings in the Switchbot app on another phone, the value changed from performance to silent and also the other way round, maybe somebody else can test this also?

How to:
I got an export of all bluetooth packets from my phone like described in https://stackoverflow.com/a/30352487 (had to use the adb shell to get a bugreport).
This export can be read by Wireshark and you can search for the mac address `bluetooth.addr==XX:XX:XX:XX:XX:XX` but I found it easier just to search for the typical switchbot magic packet with `btatt.value contains 570f`