devWaves/SwitchBot-MQTT-BLE-ESP32 #49: [Featheresp32]: Monitor battery voltage

samip5, 2022-01-28:
Do you happen to have ideas on how one would implement monitoring battery voltage from on-board JST connector as entry in HA?

Ref: https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/pinouts

devWaves, 2022-01-28:
this is not an issue. This code also probably shouldnt be used with a battery

closed 2022-01-28 by devWaves

samip5, 2022-01-28:
Why shouldnt it be uaed with a battery? It's used only as backup power for the esp.

devWaves, 2022-01-28:
battery would not last very long I am guessing, so ya backup is only if esp32 loses power. your mqtt server still needs to be working so you would need your router and HA on UPS/battery also

you can look on HA forum here for adding a sensor https://community.home-assistant.io/t/switchbot-bot-curtain-meter-contact-motion-mqtt-esp32-bridge-local-control/286468/436?u=devwaves