devWaves/SwitchBot-MQTT-BLE-ESP32 #51: [BUG] Wemos D1 Mini ESP32 Reboots (Crashes?) on commands

NickM-27, 2022-01-30:
I am using the Wemos D1 Mini ESP32 and have a curtain as well as a bot. Both exhibit similar behavior where they will have a new state set, update correctly, and then I can see on my router it disconnects and then reconnects. The MQTT state is then changed back to what it was before. If I check in HA I can see it update to the correct state, become unavailable, and then come back as the state before. 

[image]

devWaves, 2022-01-30:
can't say I have those issues.

watch the mqtt messages on switchbot/esp32 and switchbot/esp32/#

you will see what the esp32 is doing based on the mqtt messages.

If you see a "boot" message the esp32 rebooted.

make sure that no mqtt messages are retained on the 'set' topics or any topic that controls devices

NickM-27, 2022-01-30:
@devWaves Thanks! I am trying to figure out mqtt explorer but I was able to find this which seems to confirm what I thought:

[image]

NickM-27, 2022-01-30:
This is my config, I have not changed any advanced settings

```
/* Wifi Settings */
static const char* host = "switchbotController";                          //  Unique name for ESP32. The name detected by your router and MQTT. If you are using more then 1 ESPs to control different switchbots be sure to use unique hostnames. Host is the MQTT Client name and is used in MQTT topics
static const char* ssid = "REDACTED";                           //  WIFI SSID
static const char* password = "REDACTED";                   //  WIFI Password

/* MQTT Settings */
/* MQTT Client name is set to WIFI host from Wifi Settings*/
static const char* mqtt_host = "192.168.50.106";                       //  MQTT Broker server ip
static const char* mqtt_user = "REDACTED";                         //  MQTT Broker username. If empty or NULL, no authentication will be used
static const char* mqtt_pass = "REDACED";                         //  MQTT Broker password
static const int mqtt_port = 1883;                                  //  MQTT Port
static const std::string mqtt_main_topic = "switchbot";             //  MQTT main topic

/* Switchbot Bot Settings */
static std::map<std::string, std::string> allBots = {
  { "ice_maker", "f3:0f:e2:64:28:00" }
};

/* Switchbot Curtain Settings */
static const int curtainClosedPosition = 10;    // When 2 curtains are controlled (left -> right and right -> left) it's possible one of the curtains pushes one of the switchbots more open. Change this value to set a position where a curtain is still considered closed
static std::map<std::string, std::string> allCurtains = {
  { "back_door_curtain", "c4:0d:4d:58:c6:66" }
};

/* Switchbot Meter Settings */
static std::map<std::string, std::string> allMeters = {
  /*{ "meterone", "xX:xX:xX:xX:xX:xX" },
    { "metertwo", "yY:yY:yY:yY:yY:yY" }*/
};

/* Switchbot Contact Sensor Settings */
static std::map<std::string, std::string> allContactSensors = {
  /*{ "contactone", "xX:xX:xX:xX:xX:xX" },
    { "contacttwo", "yY:yY:yY:yY:yY:yY" }*/
};

/* Switchbot Motion Sensor Settings */
static std::map<std::string, std::string> allMotionSensors = {
  /*{ "motionone", "xX:xX:xX:xX:xX:xX" },
    { "motiontwo", "yY:yY:yY:yY:yY:yY" }*/
};

/* Switchbot Bot Passwords */
static std::map<std::string, std::string> allPasswords = {     // Set all the bot passwords (setup in app first). Ignore if passwords are not used
  /*{ "switchbotone", "switchbotonePassword" },
    { "switchbottwo", "switchbottwoPassword" }*/
};

/* Switchbot Bot Device Types - OPTIONAL */
/* Options include: "switch", "light", "button" */
static std::map<std::string, std::string> allBotTypes = {     // OPTIONAL - (DEFAULTS to "switch" if bot is not in list) - Will create HA entities for device types
 /* { "switchbotone", "switch" },
    { "switchbottwo", "light" },
    { "switchbotthree", "button" }*/
};

      /*** Bots in PRESS mode to simulate ON/OFF - ESP32 will try to keep track of the ON/OFF state of your device while in PRESS mode***/
      // Add bots while in PRESS mode that will simulate ON/OFF. Default state will be used if no MQTT retained on state topic
      // false = default state = OFF
      // true = default state = ON
      // If the state is incorrect, call set STATEOFF or STATEON
      static std::map<std::string, bool> botsSimulateONOFFinPRESSmode = {
        { "ice_maker", true }
      };

      //Add bots OFF hold time for simulated ON/OFF, if not in list, the current hold value will be used. Device must be in botsSimulateONOFFinPRESSmode list
      static std::map<std::string, int> botsSimulatedOFFHoldTimes = {
        /*{ "switchbotone", 0 },
          { "switchbottwo", 10 }*/
      };

      //Add bots ON hold time for simulated ON/OFF, if not in list, the current hold value will be used. Device must be in botsSimulateONOFFinPRESSmode list
      static std::map<std::string, int> botsSimulatedONHoldTimes = {
        /*{ "switchbotone", 15 },
          { "switchbottwo", 1}*/
      };
      /********************************************/
```

devWaves, 2022-01-30:
your config seems in order so not sure what the issue is

So you are using the bot with a device that doesn't have an ON/OFF correct? like a toggle button on a coffee maker or air conditioner ? otherwise you don't need botsSimulateONOFFinPRESSmode

You need to calibrate curtains with switchbot app first if you haven't. You should see a curtain topic like your bot

I can tell your bot doesn't have a password because you got the firmware value

if you see the "boot" message a lot then there is an issue somewhere with your setup that it doesn't like. You should see "boot" once when the esp32 turns on

if you only have curtain and bot you should see the blue LED on the esp32 turn off at least 2 min after booting. If it runs the full 2min it probably didn't find a device. The LED should turn off after boot once it finds all your devices. The LED will turn back on while controlling or scanning for devices.

NickM-27, 2022-01-30:
## Answers

I am using the bot with an ice maker which does have a single button to toggle on / off.

I do see a curtain topic and have calibrated them. 

I see the "boot" message come up like I posted above quite often after sending the push command to switchbot or the open/close command to the curtain.

I unplugged and replugged the esp32 and the blue stays on for 5-10 secs so it seems to be finding the devices quite quickly.

## Questions

Trying to think of what the issue could be since it does happen to the bot and the curtain (although much more often with the bot)

devWaves, 2022-01-30:
maybe try with the default names or shorter names. esp32, switchbotone, curtainone. If you said issues started with curtain, your curtain name is on the longer side

are you using Home Assistant? or manually sending commands?

maybe a simple reflash will fix it. Nothing stands out as an issue atm

NickM-27, 2022-01-31:
@devWaves I am only using it with Home Assistant, did try a few reflashes to no avail. I tried shortening the names (to "ice" and "bd") and that seems to have done it. Very interesting that the length affected it. 

I added a PR #52 to add a troubleshooting section to the README, let me know what you think.

closed 2022-01-31 by NickM-27

devWaves, 2022-02-04:
So I tried with "ice_makerXXXXXXXXXXXXXXXXX" and "back_door_curtainXXXXXXXXXX" with ice_makerXXXXXXXXXXXXXXXXX in simulated on/off mode

I had zero issues. Works for me, so there must have been something else

NickM-27, 2022-02-04:
Huh fair enough, it's been absolutely solid since so maybe I just had an unlucky few flashes and the last one I did fixed something.