devWaves/SwitchBot-MQTT-BLE-ESP32 #53: Support for Magic Switchbot

mnowok, 2022-02-01:
There is a chinese Magic switchbot similar to original one: https://thegreytechnologies.com/products/magic-switchbot-button-pusher

Communication libray and Home Assistant integration is avaialble at https://community.home-assistant.io/t/new-integration-magic-swithcbot/291972

I would be awsome if SwitchBot-MQTT-BLE-ESP32 project could suport these devices. I have both types of switchbots and  maybe some other users also would benefit from this feature.

devWaves, 2022-02-04:
Magic switchbot works differently than switchbot and I believe it uses more encryption. It would take significant work to support them and I don't own any so this probably won't happen

If you are using HA, there is an HA integration for them https://community.home-assistant.io/t/new-integration-magic-swithcbot/291972

mnowok, 2022-02-05:
Ok, it was just a suggestion. I understand your explanation, Of course I'm using HACS integration, but as I had some serious problems with HA original switchbot integration I found your ESP solution. I switched original switchot to ESP and I just thought it would be great to use ESP for both kind of devices. HACS magic switchbot integration isn't actively developed.

Anyway thanks for your feedback, Using SwitchBot-MQTT-BLE-ESP32 and HACS Magic Switchbot seems to be stable for now.