devWaves/SwitchBot-MQTT-BLE-ESP32 #55: Becoming unavailable every ~24 hours

ScottG489, 2022-02-18:
Every ~24 hours my ESP32 device becomes unavailable. I don't know if it's freezing or dropping off the network or what. Replugging it back in, it immediately becomes available again and works great (until it drops off again). I've tried a few different power sources in case something where was going on there, but no luck.

This is the ESP32 device I'm installed on:

[HiLetgo ESP-WROOM-32 ESP32 ESP-32S Development Board 2.4GHz Dual-Mode WiFi + Bluetooth Dual Cores Microcontroller Processor Integrated with Antenna RF AMP Filter AP STA for Arduino IDE](https://www.amazon.com/gp/product/B0718T232Z/
)

Do you have any idea what this may be or if there's a way I could debug the device to find out what could be going wrong?

devWaves, 2022-02-18:
I am currently testing this. There might be a bug in the NimBLE library. I noticed issues with the recent versions.

I am trying to see if 1.3.3 NimBLE makes things stable

thetic, 2022-02-18:
I see this happen too, but less frequently (every few days maybe). I have 2 switchbot curtains connected.

devWaves, 2022-02-19:
I am thinking that version 1.3.3 is stable. I am at 48h now, still running no issues. There are a couple bugs that came up recently in the NimBLE github issues

I will continue to test, but so far so good with 1.3.3

FYI: I have like 2 of every device type on one ESP32 (2 bots, 2 curtain, 2 meter, 2 contact, 2 motion)

[image]

ScottG489, 2022-02-20:
I only have two SwitchBot Bot's (S1). No motion or contact sensors.

ScottG489, 2022-02-20:
Here's the history of the up/downtime of my ESP32 device (red being down):
[image]

devWaves, 2022-02-21:
Can you tell me which version of NimBLE-Arduino you are using? there are definitely some recent issues that came up with that library. I was able to get a failure in the recent versions, but I am still running 1.3.3 at like 82 h now no issue yet

until I figure out the exact issue though if you want to apply a workaround, this code will reboot the esp32 every X number of hours. I have it set to 12 hours now. Just add it to the existing code...

```
int howOftenToReboot = 12;   // Every X hours the ESP32 will reboot

void loop() {
  if (millis() > (howOftenToReboot * 1000 * 60 * 60 )) {
    ESP.restart();
  }

  /*
    current loop code
  */

}
```

ScottG489, 2022-02-22:
For sure. How can I find out what version of NimBLE-Arduino I'm on?

devWaves, 2022-02-22:
@ScottG489 are u using arduino? if yes the library version is in the same spot you installed it

Sketch > Include Library > Manage Libraries

screenshot here https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/issues/55#issuecomment-1046065066

ScottG489, 2022-02-23:
I used VS Code to do the install. I had tried Arduino IDE but ran into some issues. Do you happen to know how to check the version in VS Code? I'll be able to poke around for myself soon, but I haven't had access to the machine where I flashed this firmware to my ESP32 device.

FWIW I did the flashing around Feb 8th.

devWaves, 2022-02-25:
Ya I am pretty sure 1.3.7 should solve the issue. I am a like 5 days now no issues. I did experience the issue in previous Nimble versions and they were easy enough to reproduce after a day or 2. 

Please install 1.3.7 Nimble and let me know

ScottG489, 2022-02-25:
Ok thanks for looking into this. I should be able to get to this this weekend and I'll let you know how it goes.

ScottG489, 2022-02-28:
OK it seems like I'm on `1.3.6`. I'll reinstall with a `1.3.7`.

I'll report back here on how the devices is fairing over the next few days.

ScottG489, 2022-03-03:
Been running strong since Sunday.

@devWaves Is there anything on your end that you need to do for this fix to be implemented?

Otherwise, if I don't have any other issues, I'll close this ticket this weekend.

devWaves, 2022-03-03:
hey @ScottG489 . Thanks for confirming and glad that fixed the issues you were seeing.

I've updated the VS code to pull version 1.3.7. But no other changes for my stuff.

I'll close the issue.

@thetic if u can let me know if 1.3.7 solved the issues u see that would be good. thx

closed 2022-03-03 by devWaves

calvolson, 2022-06-27:
This appears to be an issue with 1.3.8 as well. After downgrading to 1.3.7 things look to be working great.

brandon-daniels, 2023-01-15:
I don't know if anyone else is having this problem, but I have this exact same issue and I have the exact libraries recommended by the installation guide:
NimBLE-Arduino library version 1.4.0 (most recent is 1.4.1) 
EspMQTTClient library version 1.13.3
ArduinoJson library version 6.19.4
CRC32 library (by Christopher Baker version 2.0.0
ArduinoQueue librar version 1.2.5

I'm running this one the ESP32 Dev Module https://www.amazon.com/dp/B07WCG1PLV?ref=ppx_yo2ov_dt_b_product_details&th=1