devWaves/SwitchBot-MQTT-BLE-ESP32 #60: Battery Status as a separate subject

Trrw, 2022-03-25:
Hello,

Would it be possible to send the battery status as a separate subject?
Also for example /switchbot/esp32/curtain/curtainone/battery

The value is currently stored in the attribute topic with json.

Thank you!

Trrw, 2022-03-28:
Thank you for the work! Would it also be possible to create a own subject (to controll) for the position of the Switchbot curtain?

devWaves, 2022-03-28:
You can already set the position by sending a number between 1-100 to the curtain set topic

mclever, 2022-05-11:
Ran into an issue today where last battery update to HA showed 55% but the battery was actually dead. Batt % went right up to 100 after replacement so no sure what happened

mclever, 2022-07-01:
Forgot about updating since everything is working so great. Thanks again for this awesome solution!

Had one of my switchbots stop fingering at 55% battery today and was still sending link quality. I'll probably just set an automation to alert at battery =55% for next time since I'm reading about all the BLE issues.

devWaves, 2022-08-09:
v7.0 release and adds topics for battery and rssi as well as the included battery hack to send 1% battery is devices respond with low battery warning on command attempt

closed 2022-08-09 by devWaves