devWaves/SwitchBot-MQTT-BLE-ESP32 #61: Delay in actions

Ulrar, 2022-04-15:
Hi,

I just set this up on a Doit dev kit v1, and when I press the buttons in HA the delay between the button press and the switchbot curtains actually moving is very inconsistent. 
Sometimes it goes right away but it usually takes a while, I've seen it take up to 15 seconds.

Is there a way to figure out why it's that slow ? I have the board a couple of meters away from the switchbots, I can't get it any closer. Rssi is somewhere between 50 and 70 usually. 

The blue LED on the board comes on when the curtains are moving, not when the button is pressed. Does that mean the delay is in mqtt or in ble ?

Thanks

Ulrar, 2022-04-16:
Looking at the comments around the LED defines, it sounds like the blue LED comes on when it's processing a command from MQTT.

So presumably the fact that the LED is on while the curtains start moving would indicate that the link between the switchbots and the board is fine, but for some reason there's a huge delay somewhere in the chain HA -> MQTT -> ESP32.
HA and MQTT are running on a proper x86_64 server, not a pi so it's probably not just the server being slow, but is there a way to make sure ?

devWaves, 2022-04-16:
when the esp32 receives and starts processing the command you will see ```{status":"controlling"}``` on the main MQTT topic. This should happen basically immediately

The LED should stay on for entire duration from the time the command is received to when it was sent successfully. If the LED is not on for the full 15 seconds then the delay would seems to be from HA MQTT

you can install the MQTTExplorer app to easily watch all the mqtt traffic. The esp32 status should change to controlling right after the MQTT message is received on the 'set' topic

if you plug the esp32 into your pc you can watch the serial monitor messages by changing this value in the code ```printSerialOutputForDebugging = true;```

Ulrar, 2022-04-17:
Hey,

Thanks for that, tried it with the debug on and using `mosquitto_sub -h 127.0.0.1 -p 1883 -u mqttuser -P mqttpassword -t "switchbotrelay-livingroom/#" -d` to monitor the mqtt topics.

When I click on the button in home assisstant, I can see the mqtt message imediatly :
```
Client mosq/Kutw5gau7ld3sge6bv received PUBLISH (d0, q0, r0, m0, 'switchbotrelay-livingroom/esp32/curtain/main-door/set', ... (5 bytes))
```
But the esp32's logs nothing at that time. Maybe 10 seconds later, the esp32 logged the command and the switchbot immediatly started moving.

So I guess the issue is between MQTT and the esp32, any ideas why it sometimes doesn't get the message right away ?
It's got decent signal, according to the unifi controller the `experience` is 75% whatever that means, and HA reports the esp's rssi at around -65.

Ulrar, 2022-04-17:
Correction, it seems fixed now. I just created a separate user for HA and for the ESP32 and reconfigured home assistant, not sure which of these two actions solved the issue but it just works now. I wonder if sharing a user was causing it ?
Anyway I'll keep a close eye on it for a day or two and if the issue doesn't return I'll close it, seems like it was my setup indeed.
Thanks !

Ulrar, 2022-04-18:
Still working perfectly fine, must have been the shared user.
Closing it then, thanks !

closed 2022-04-18 by Ulrar