devWaves/SwitchBot-MQTT-BLE-ESP32 #62: [Feature request] Monitoring support

thematrixdev, 2022-04-23:
Hello there.

I have found the ESP32 stops working periodically.
I think it is due to WIFI connectivity.
May I request to add a feature, to PING a given URL periodically, which the `URL` and `INTERVAL` could be specified in settings?
I am using healthchecks.io as service monitor, and via webhook to instruct HomeAssistant to reset the power of the ESP32.

Thanks.

devWaves, 2022-04-23:
are u running all the latest libraries? there was an issue before with libraries but that should be resolved

thematrixdev, 2022-04-23:
Should be. I believe it is due to Zigbee and Wifi issue at my home.
Health monitoring should be good anyway.
Hope you can add this feature if you have time.
Thanks.

devWaves, 2022-04-23:
I am running a ton of zigbee, bluetooth, and wifi stuff. Just make sure your zigbee channel doesnt overlap with your wifi channel. I have zero issues

double check you are running 1.3.7 of the nimble library

Its probably not a feature I would be looking to add since the problem may be somewhere else. I am also trying to figure out an issue before I release v7.

The esp32 wifi is handled by the MQTT library so it should be reconnecting

If the esp32 is not responding then it means it probably wouldnt run the ping code either. If the esp32 is working properly it should reconnect to wifi

closed 2022-04-23 by thematrixdev