devWaves/SwitchBot-MQTT-BLE-ESP32 #63: Adding contact sensor breaks curtains

Ulrar, 2022-04-28:
Hi,

I've been using this to control a pair of curtain bots for a few weeks and it's been working mostly okay, except some unexplained delays here and there.

I just added a contact sensor, and the blue LED is now just always on. The contact sensor is working great in HA and updates quickly, but the curtains are now unresponsive.
Is there some conflict there, is the connection to the contact sensor left on all the time and preventing connections to other devices ?

This is what is looping on the output : 

```
Adding Our Service ...
switchbotContactLivingRoomSideDoor
callForInfoAdvDev
Assigned advDevService
// The below repeats, a lot
Advertised Device found: Name: , Address: <some mac>, manufacturer data: ..., serviceUUID: ...
Service Data:
UUID: 0x0d00, Data: cп`2
[...]
//
Adding Our Service ...
switchbotContactLivingRoomSideDoor
callForInfoAdvDev
Assigned advDevService
``` 

I shorten it, but the mac(s) it finds there aren't related, likely some other devices somewhere in the house.
Left it for a few minutes and it looks like that's all that repeats. It connects to the curtains once at the very start, but as soon as it connects to the contact sensor that's all it outputs, endlessly.

Thanks

devWaves, 2022-04-28:
I am using curtain and contact sensor and works fine for me

the LED will stay on all the time because the contact and motiom sensor require the esp32 to scan all the time. If u only had curtains, the LED turns off because scanning is only done when needed

the output u see is correct

Ulrar, 2022-04-28:
Right, but in my case as soon as I add the contact sensor the curtains stop working.
Any idea what I can do ?

devWaves, 2022-04-28:
I see your device names and your switchbot topic are very long. I've tried with long names and didnt have issues, but it has been brought up before

I would suggest trying with the default names.

Otherwise I would have to look at the code u used

Ulrar, 2022-04-28:
I did change that since, but you're right I did have very long names, just shortened them down a lot.
Unfortunately it's not behaving any better, although I did get one curtain command to go through after a lot of tries. Can't get it to work again with debug on.

Noticed this in the logs when sending a (failed) command to the curtain : 
```
Received something on switchbot/lr/control
main-door
Processing Control MQTT...
Device: main-door
Device value: CLOSE
Try to connect. Try a reconnect first...
New client created
Connected
Connected to: ca:db:00:8e:83:69
RSSI: -58
Sending command...
subscribed to notify
Wrote new value to: cba20002-224d-11e6-9fb8-0002a5d5c51b
notifyCB
deviceName: WoCurtain
Success! Command sent/received to/from SwitchBot
The response value from curtain: 14
unsubscribed to notify
Done sendCommand...
```

That makes it sound like the command was received successfully, even though the curtains did not close.
Is 14 an expected return value ? In HA the icon changes to closed then back to open on it's own, presumably on the next update.

I've also tried it with the default hostname and topic just now, same behavior.
As for the codes I just looked at git diff to make sure and the only changes from the visualstudio folder are : 
- WiFi info
- MQTT Info
- Curtains closed position to 15
- One curtain "curtainone" in allCurtains
- One contact "contactone" in allContactSensors
- The log flag to true when debugging, false the rest of the time
It's compiled for a esp32doit-devkit-v1.

devWaves, 2022-04-29:
ya you will want to get to get ot workomg with the switchbot app. sounds like a problem with the curtain itself

from the switchbot python host documentation the error code 14 = 0x0E - Disconnected from the device that needs to stay connected..... whatever that means

Ulrar, 2022-04-29:
Yeah so there was something weird going on with the curtains again. I charged them overnight and they weren't working this morning still, so I had to go through the rest procedure again (45 seconds on the button for each, then re-add and re-calibrate in the app) and it's fixed.

Not sure what's going on but for some reason the curtains break when I also add the contact sensor so let's just not. It's just BLE advertising so I used esphome to expose the contact sensors instead without an active connection and that works fine, and doesn't seem to conflict with the curtains.

This will sadly remain unexplained, but let's just close it I guess. Thanks !

closed 2022-04-29 by Ulrar

devWaves, 2022-04-29:
I just found an issue when upgrading to NimBLE library version 1.3.8. Not sure if it is the same issue, but curtains work for one control then the esp32 freezes

NimBLE 1.3.7 works. I am going to create some issue report for this

Ulrar, 2022-05-01:
My platformio.ini file says `h2zero/NimBLE-Arduino@^1.3.7` so presumably that's not it :(