devWaves/SwitchBot-MQTT-BLE-ESP32 #64: NimBLE 1.3.8 breaks functionality

devWaves, 2022-04-29:
I just found out that when installing NimBLE 1.3.8 it breaks stuff. Not sure what the issue is but stick with 1.3.7 for now

Jaxom99, 2022-07-19:
Thanks for notifiying this, I found myself in the same situation. 
Did you open a ticket on their side ? Like in this repo perhaps : https://github.com/h2zero/NimBLE-Arduino

devWaves, 2022-07-19:
@Jaxom99 I haven't had the chance to look if I can fix it from my end yet. I was going to do that once I had more info to provide or some sample basic code to simulate the issue. To reproduce thr issue with current code, they would need switchbot devices

devWaves, 2022-07-29:
I've got a fix for the NimBLE issues. I will release a v6.12 with the fix. Just want to test a bit more

devWaves, 2022-07-30:
v6.12 released and solves the NimBLE v1.3.8, v1.4.0 issues

closed 2022-07-30 by devWaves

Jaxom99, 2022-07-31:
thanks, I will test it soon as well!