devWaves/SwitchBot-MQTT-BLE-ESP32 #67: errorConnect with switchbots bot

baty0man, 2022-05-06:
Hi,

I'm pretty new with MTQQ,. I've set up the ESP32 board and it can read the battery of my switchbot bots successfully and the entities appear on HA. However when I toggle the switch ion HA it doesn't do anything. Looking at the messages on the status topic i get the following:

{"status":"errorConnect","command":"OFF"}

Not sure what I'm doing wrong, any help would be appreciated.

Thanks

devWaves, 2022-05-06:
Is your esp32 close enough to the bot? what is the linkquality rssi value that shows in HA?

baty0man, 2022-05-08:
Sorry for the late reply, the esp32 board is about a meter away from the bots.
[image]

devWaves, 2022-05-11:
I have just made updates to a v7.0-preRelease here
https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/tree/v7alpha

Can you try the v7.0-preRelease and let me know if you still have issues?

just ignore the new settings related to "mesh"

baty0man, 2022-05-12:
Still the same issue unfortunately. I've deleted all topics and HA entities, uploaded the new version. I can see the new entities on HA but when I press the switch button, nothing happens:
[image]

devWaves, 2022-05-12:
hmm.. nothing stands out. device names look fine. I can see the guestroom bot worked to request settings, but not livingroom bot

Are any other bluetooth things trying to connect to the bots like switchbot hub etc? Maybe try taking a bot battery out, put it back then turn on the esp32

I am running firmware 6.2 for bots, but 4.9 is fine also

I can try with your device names later, but it should work

Jacksonbm1, 2022-06-02:
I also have been trying to use the switchbot bots. They are autodiscovered by homeassistant, but controlling them seems like it's not working. I'm not super familiar with homeassistant or MQTT, so this might be tough for me to help you debug, but lmk what would be helpful. I've treid removing the battery and putting it back in. I do have other devices attempting to connect, but since it receives BLE commands, shouldn't it not matter?

Jacksonbm1, 2022-06-02:
Also, when I flip the switch on homeassistant, it looks like it's trying to push a new attribute in the JSON: 

[image]

note that in the "Desk_up" section, I've pressed the homeassitant button and it's added a "set = ON" to the MQTT explorer output

Jacksonbm1, 2022-06-02:
Update it was: https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/issues/64

devWaves, 2022-06-04:
@baty0man is it still giving you issues?  Are you using NimBLE v 1.3.7 as mentioned? There is a known issue when using v1.3.8

devWaves, 2022-07-03:
Closing this as I believe NimBLE v1.3.7 solves it. Can be reopened if still an issue

closed 2022-07-03 by devWaves