devWaves/SwitchBot-MQTT-BLE-ESP32 #69: Second button not updating entities - error: "Platform mqtt does not generate unique IDs"

ScottG489, 2022-05-09:
I've been getting errors like this since I think my first flash.

> Platform mqtt does not generate unique IDs. ID switchbot_\<MAC\>_linkquality already exists - ignoring sensor.lower_bed_switchbot_linkquality

I have two buttons. One is not getting these errors or experiencing any problems with it's associated entities.

The other button itself works fine, but all the entities associated with it like **Battery**, **HoldSecs**, **Linkquality**, etc. aren't updating.

Here is some relevant settings code in case I've missconfigured something:
```cpp
TBD
```

Any idea what might be going on?

devWaves, 2022-05-11:
I have just made updates to a v7.0-preRelease here
https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/tree/v7alpha

Can you try the v7.0-preRelease and let me know if you still have issues?

just ignore the new settings related to "mesh"

ScottG489, 2022-05-11:
Do you have any reason to suspect it's actually caused by this firmware?

I reinstalled my mosquitto broker and the errors went away. I think it was due to something in my mosquitto.db not clearing properly. Otherwise, I can close this issue.

devWaves, 2022-05-11:
all good then. don't need to update if it's working

there is a minor change for bots, if they return a low battery error when attempting to control, a 1% battery will be sent on the new battery topic

clearing mqtt retained topics fixes most issues. the mqtt exploret program is good for that. that probably happened when u reinstalled mqtt

closed 2022-05-11 by devWaves