devWaves/SwitchBot-MQTT-BLE-ESP32 #73: no SwitchBot action from HA triggers

m3obsessed, 2022-06-01:
@devWaves
Please help!  But first some context.  I have had 4 switchbots in HA, thanks to the Switchbot integration.  As I continue to have issues (i.e., sensors often unavailable), I decided to try the Switchbot-MQTT-BLE-ESP32 route.  No issue downloading the code, editing for wifi, MQTT user name, etc... compiling and uploading to my D1 Mini.  I can see all 4 switchbots with MQTT explorer.  And I do see them within the MQTT Integration within in HA - see below.

[image]

When I select one of the sensors (blinds_down), I can see all the right entities.

[image]

Unfortunately, when I try to activate the switchbot by pressing the switch, nothing happens with the Switchbot.
   
[image]

Honestly, I am lost (and stuck).  Not sure what to do....  Any ideas?
Thanks in advance for any help/perspective you can provide.
Gerard

devWaves, 2022-06-01:
Are you using v1.3.8 of NimBLE library? if so switch to v1.3.7. Something changed in 1.3.8 that broke stuff

If you watch the main MQTT topic it should give you some info of whats happening. If u can provide some MQTT messages it will say if it failed or success

m3obsessed, 2022-06-02:
Thanks for getting back to me so quickly.  I was using v1.3.8 of NimBLE library.  I switched to v1.3.7.  Recompiled and re-uploaded to my D1 Mini ESP32.  Still no action from the switchbot when I turn it on/off within HA.
How would you suggest I watch the main MQTT topic such that I can share some of the MQTT messages (sorry - this is beyond my current knowledge/capability)?
Gerard

m3obsessed, 2022-06-02:
@devWaves From within Home Assistant / MQTT Integration, I listened to Topic:  switchbot/D1_Mini_esp32/bot/Blinds_Stop/set

I activated the switch button three times.  No action from the bot.  But this is what I got:
[image]
 
Gerard

devWaves, 2022-06-04:
If you are using MQTTExplorer it is easy to see the MQTT traffic as it goes by

The set message would be sent by HA to the MQTT broker, so even with the ESP32 off you would see that. You would want to find a message sent by the ESP32 to diagnose more (or you can also plug it into your PC and use serial monitor to watch the logs)

As soon as a set message is received you should see the main MQTT topic  (for you ```switchbot/D1_Mini_esp32```) show ```{"status":"controlling"}```. This means the ESP32 received the command and is trying to process it

Once the ESP32 receives the command you should see messages on the device topic ```switchbot/D1_Mini_esp32/bot/Blinds_Stop/status```  that will have messages similar to this ....

```
Example payload:
                      - {"status":"connected", "command":"ON"}
                      - {"status":"errorConnect", "command":"ON"}
                      - {"status":"errorCommand", "command":"NOTVALID"}
                      - {"status":"commandSent", "command":"ON"}
                      - {"status":"busy", "value":3, "command":"ON"}
                      - {"status":"failed", "value":9, "command":"ON"}
                      - {"status":"success", "value":1, "command":"ON"}
                      - {"status":"success", "value":5, "command":"PRESS"}
                      ```

If none of those messages are displayed then the issue might be that the ESP32 can't read/subscribe to the set topic? 
Are you using an MQTT user/password that is used by other stuff? Maybe create a unique MQTT user for the ESP32. Someone else mentioned they had issues sharing MQTT userIds

m3obsessed, 2022-06-06:
@devWaves 
Thank you for your guidance.  So I looked at the main MQTT topic "switchbot/D1_Mini_esp32".  It showed either "scanning" or "idle".  I looked at the history, that is all I could see.
[image]
 
Then I noticed that the signal was not as high as what you had recommended.  So I moved the ESP32 close to the Raspberry Pi running HA and the four Switchbots.  Left it unplugged while I deleted all Switchbot devices in HA, and deleted all the ESP32-related topics in MQTT Explorer.  

I then plugged the D1 Mini back in.  Let MQT rediscover them.  Went to MQTT Explorer....  AND MAGIC!!!!
[image]

ESP32 is controlling.  All four Switchbots are showing "success".  I tried them all from within HA and they all work.

At the end, doing some cleanup, moving the ESP32 closer to the Switchbots did the trick.  

Anyway, thank you SO MUCH for your help.  I learned a few things about MQTT Explorer in the process.  And THANK YOU for making this really cool integration between Switchbot and HA via ESP32....  Really cool stuff.

Gerard

devWaves, 2022-06-07:
awesome glad you got it working. Just a npte that it doesnt really matter how close the esp32 is to the RPi, just needs to be close to the switchbots. The wifi signal can work from really far away, it os the bluetooth signal that needs to be good

have a good one, I will close the issue

closed 2022-06-07 by devWaves