devWaves/SwitchBot-MQTT-BLE-ESP32 #75: Flash Error:  static byte bArray...

Trrw, 2022-06-04:
Hello,

I would like to flash a new ESP32. Now I become this Error:

```
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:695:69: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayPos[] =  {0x57, 0x0F, 0x45, 0x01, 0x05, 0xFF, NULL};
                                                                     ^
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:697:56: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayHoldSecs[] = {0x57, 0x0F, 0x08, NULL };
                                                        ^
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:698:60: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayBotMode[] = {0x57, 0x03, 0x64, NULL, NULL};
                                                            ^
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:698:60: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:700:68: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayPressPass[] = {0x57, 0x11, NULL, NULL, NULL, NULL};
                                                                    ^
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:700:68: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:700:68: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:700:68: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:701:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayOnPass[] = {0x57, 0x11, NULL , NULL, NULL, NULL, 0x01};
                                                                        ^
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:701:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:701:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:701:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:702:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayOffPass[] = {0x57, 0x11, NULL, NULL, NULL, NULL, 0x02};
                                                                        ^
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:702:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:702:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:702:72: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:703:74: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayGetSettingsPass[] = {0x57, 0x12, NULL, NULL, NULL, NULL};
                                                                          ^
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:703:74: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:703:74: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:703:74: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:704:84: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayHoldSecsPass[] = {0x57, 0x1F, NULL, NULL, NULL, NULL, 0x08, NULL };
                                                                                    ^
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:704:84: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:704:84: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:704:84: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:704:84: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:705:82: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayBotModePass[] = {0x57, 0x13, 0x64, NULL, NULL, NULL, NULL, NULL};     // Other github documentation shows this to be the array for setting mode with password (firmware 4.5, 4.6)
                                                                                  ^
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:705:82: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:705:82: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:705:82: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:705:82: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
[... cut ...]
 static byte bArrayBotModePass[] = {0x57, 0x13, NULL, NULL, NULL, NULL, 0x64, NULL};       // The proper array to use for setting mode with password (firmware 4.9)
                               ^
/Users/User/Downloads/SwitchBot-MQTT-BLE-ESP32-main/Arduino IDE Files/SwitchBot-BLE2MQTT-ESP32/SwitchBot-BLE2MQTT-ESP32.ino:705:13: note: 'byte bArrayBotModePass [8]' previously defined here
 static byte bArrayBotModePass[] = {0x57, 0x13, 0x64, NULL, NULL, NULL, NULL, NULL};     // Other github documentation shows this to be the array for setting mode with password (firmware 4.5, 4.6)
```

What can I do? Thank you!

Trrw, 2022-06-04:
The ESP32 is flashed with the errors. I become the mqqt data. But I can't control the Curtain Bot with mqtt command

devWaves, 2022-06-04:
You should not be getting any errors. You will get warnings in PlatformIO, but they can be ignored

this part of the code is commented out
```
//static byte bArrayBotModePass[] = {0x57, 0x13, 0x64, NULL, NULL, NULL, NULL, NULL};     // Other github documentation shows this to be the array for setting mode with password (firmware 4.5, 4.6)```

so you should not be getting this error...
```
SwitchBot-BLE2MQTT-ESP32:706:31: error: redefinition of 'byte bArrayBotModePass []'
 static byte bArrayBotModePass[] = {0x57, 0x13, NULL, NULL, NULL, NULL, 0x64, NULL};       // The proper array to use for setting mode with password (firmware 4.9)
```

If you use Arduino IDE for other projects, you might want to uninstall, delete the Arduino libraries from C:\Users\<userId>\Documents\Arduino\libraries and then start fresh

What version of the curtain firmware do you have? I have v3.9 curtain firmware

If you previously had it working and changed device names or MQTT topics, ensure to delete the previous MQTT topics for the switchbot topics and the homeassistant discovery topics. You can use the program MQTT explorer for that easily

fantnhu, 2022-07-02:
I get the same error. Fresh Arduino installation:
The line mentioned above is commented out. Only 2pcs SwitchBot ROD2.
No need to deal with it?
```
...
C:\Users\pinte\OneDrive\Dokumentumok\Arduino\SwitchBot-MQTT-BLE-ESP32\SwitchBot-MQTT-BLE-ESP32.ino:702:84: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
C:\Users\pinte\OneDrive\Dokumentumok\Arduino\SwitchBot-MQTT-BLE-ESP32\SwitchBot-MQTT-BLE-ESP32.ino:702:84: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
C:\Users\pinte\OneDrive\Dokumentumok\Arduino\SwitchBot-MQTT-BLE-ESP32\SwitchBot-MQTT-BLE-ESP32.ino:702:84: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
C:\Users\pinte\OneDrive\Dokumentumok\Arduino\SwitchBot-MQTT-BLE-ESP32\SwitchBot-MQTT-BLE-ESP32.ino:702:84: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
C:\Users\pinte\OneDrive\Dokumentumok\Arduino\SwitchBot-MQTT-BLE-ESP32\SwitchBot-MQTT-BLE-ESP32.ino:704:82: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
 static byte bArrayBotModePass[] = {0x57, 0x13, NULL, NULL, NULL, NULL, 0x64, NULL};       // The proper array to use for setting mode with password (firmware 4.9)
                                                                                  ^
C:\Users\pinte\OneDrive\Dokumentumok\Arduino\SwitchBot-MQTT-BLE-ESP32\SwitchBot-MQTT-BLE-ESP32.ino:704:82: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
C:\Users\pinte\OneDrive\Dokumentumok\Arduino\SwitchBot-MQTT-BLE-ESP32\SwitchBot-MQTT-BLE-ESP32.ino:704:82: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
C:\Users\pinte\OneDrive\Dokumentumok\Arduino\SwitchBot-MQTT-BLE-ESP32\SwitchBot-MQTT-BLE-ESP32.ino:704:82: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]
C:\Users\pinte\OneDrive\Dokumentumok\Arduino\SwitchBot-MQTT-BLE-ESP32\SwitchBot-MQTT-BLE-ESP32.ino:704:82: warning: converting to non-pointer type 'byte' {aka 'unsigned char'} from NULL [-Wconversion-null]

```

fantnhu, 2022-07-02:
> Az ESP32 villog a hibákkal. Én leszek az mqqt adat. De nem tudom irányítani a Curtain Bot-ot mqtt paranccsal

The device appears perfectly in HomeAssistant, I can see the entities, but I can't control it either. (ROD2)

devWaves, 2022-07-02:
these are warning, not errors. They can be ignored

devWaves, 2022-07-02:
@fantnhu from your other screenshot u have an rssi of -97 which is not a good enough signal to control. u need probably a minimum of -80. place the esp32 close to the curtain and try again

fantnhu, 2022-07-02:
> @fantnhuA másik képernyőképedből -97-es rssi-je van, ami nem elég jó jel a vezérléshez. valószínűleg minimum -80 kell. helyezze az esp32-t a függöny közelébe, és próbálja újra

Yeah, that's weird to me too. The device is roughly 2m away, but I will try to put it next to it with a longer USB cable. If it works like that, I'll buy an antenna.

devWaves, 2022-07-02:
I am not 100% positive the rod2 uses the same commands as the 1st version, but I assume they do

you can test sending curtain commands using the nRF connect app with a smartphone and these instructions https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/issues/48#issuecomment-1004152003

make sure to be standing close to the curtain when testing

fantnhu, 2022-07-02:
> Nem vagyok 100%-ig biztos benne, hogy a rod2 ugyanazokat a parancsokat használja, mint az 1. verzió, de feltételezem, hogy igen
> 
> tesztelheti a függönyparancsok küldését az nRF connect alkalmazással okostelefonnal, és ezekkel az utasításokkal [#48 (megjegyzés)](https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/issues/48#issuecomment-1004152003)
> 
> győződjön meg arról, hogy a függöny közelében áll a tesztelés során

Thanks, I know the nRF program. I will test it from home and send a picture of it.

devWaves, 2022-07-02:
> Yeah, that's weird to me too. The device is roughly 2m away, but I will try to put it next to it with a longer USB cable. If it works like that, I'll buy an antenna.

if you bought the (ESP32-WROOM-32U) this requires an antenna. It does not have a built in antenna like other esp32 model. If you aren't using an antenna now this is definitely the issue

fantnhu, 2022-07-02:
> > > @fantnhuA másik képernyőképedből -97-es rssi-je van, ami nem elég jó jel a vezérléshez. minimális minimum -80 kell. helyezze az esp32-t a függöny közelébe, és próbálja újra
> > 
> > 
> > Igen, ez nekem is furcsa. A készülék nagyjából 2m-re van, de megpróbálom hosszabb USB kábellel mellé rakni. Ha így működik, veszek egy antennát.
> 
> ha az (ESP32-WROOM-32U) készüléket vásárolta, ehhez antenna kell. Nincs beépített antennája, mint a többi esp32 modellnek. Ha most nem használ antennát, akkor biztosan ez a probléma

OK, I'll give it a go. I use several of the same ESP32 at home for Bluetooth devices without an antenna. No problem there.
I use 6pcs Xiaomi thermometers with it. There is no problem with it in a remote part of the apartment. I'll try it up close, if it's not good, I'll buy an antenna. :)

devWaves, 2022-07-03:
@Trrw  if you are running curtain v4.6 this would be the reason. I just added a fix in v6.11

Any warnings you see can be ignored. Most are because I put NULL in a char array, but the NULLs get replace in the code

@fantnhu if you still have issues you can open a new ticket, but as mentioned, for you the rssi is too low and most likely the issue

closed 2022-07-03 by devWaves

fantnhu, 2022-07-03:
Hi! I tried it with nRF, it's really the same command.
And you were right, an antenna solved the problem.
Thank you for your help and patience!

devWaves, 2022-07-03:
> Hi! I tried it with nRF, it's really the same command. And you were right, an antenna solved the problem. Thank you for your help and patience!

no problem @fantnhu . Glad you got it working. If you were using the esp32-wroom-32u with other stuff without an antenna, adding an antenna will give much better performance as there is no built in antenna