devWaves/SwitchBot-MQTT-BLE-ESP32 #78: Switch Bot Mini Hub support

pmcdermid, 2022-06-21:
Just configured this for 3 door sensors (now planning to get more since i can integrate into HA better) and a temperature sensor.  Would be great if we could then interact with the hub and various devices that have been added for example I have added my Air Conditioner and want to be able to say in HA
WHEN  door sensor closed
ON week days
AND Window Sensore is closed
AND Temperature above 26C 
Turn On AC.

devWaves, 2022-06-23:
I don't believe the hub supports BLE commands and can only be called from WiFi/internet/api

there also isnt documentation which makes things difficult for the BLE commands to send (if supported). reading BLE data is easy enough to reverse engineer but thats not what is needed here

pmcdermid, 2022-06-24:
Thanks for replying @devWaves - I wasn't sure what options were available, i saw that it had a BLE MAC as well as its WI-FI but I guess that might just beto connect to the switch bot devices devices. I might go back and see if i can get the switchbot integration in HA to work and see if that offers any control for hub devices

devWaves, 2022-06-29:
closing for now, but can be reopened if hub ever gets documentation for BLE control support

closed 2022-06-29 by devWaves