devWaves/SwitchBot-MQTT-BLE-ESP32 #79: SwitchBot Lock Support

quagmirert, 2022-06-21:
Just configured this and it's working great. Currently  have a SwitchBot Lock and would really like to integrate this into my Home Assistant and would be willing to help test and debug as necessary.

nitro41992, 2022-06-25:
I second this! I saw that it definitely does use encryption for security purposes.

What kind of documentation is required, just for my education?

I would also be willing to help debug/test if this request is taken on

devWaves, 2022-06-25:
the current request for documentation is here [https://github.com/OpenWonderLabs/SwitchBotAPI-BLE/issues/5](lock documentation request)

Jacksonbm1, 2022-07-16:
I saw a post in a live chat with a cofounder of switchbot that said the ble lock api is coming:

Here's the op:

"Hey. It's Richard here. I am one of the cofounders of
this little robot SwitchBot. I am more on sales and
marketing, but i also know how to code and play
around with JavaScript and Python myself. I have
around 160+ connected device, 20-30 scenes/
routines/automations or whatever you call it at my
place and most of them are in use, which means I
kinda know what smart home is. Also I use Alexa
most of the time at home and Siri when I am on my
phone. Just wanna say hi and if you guys have
anything you wanna tell me, just let me know. Would
like to interact with you guys more from now on. In
case i missed you comments, pin me again. I am
also on Twitter in case you wanna talk there too.
https://twitter.com/richardmou63/"

Here's the reply that says the lock api is coming:

"1. Lock API is definitely on its way. We gotta
test the security level before releasing and
there is no ETA yet."

Post: https://www.reddit.com/r/TrySwitchBot/comments/vzr3y6/first_post_hello_world/?utm_source=share&utm_medium=ios_app&utm_name=iossmf

Jacksonbm1, 2022-07-16:
Also, just for the sake of clarity, I believe [this](https://github.com/OpenWonderLabs/node-switchbot/issues/90) is the big that is blocking this bug.

Jacksonbm1, 2022-07-19:
Nope. They seemed to have removed their response, but the question explicitly stated ble "Will switchbot be releasing their ble api for the locks? The hub api is out and also apis for the bots and stuff."

Vorms, 2022-08-31:
Hello, i buy a lock and a mini hub.
The publicity tell devices can be accessed from outside home . It isn't I see the gateway project. This can be realy helpfull if it will support the lock. So keep me in touch! I will be really impressed! Best regards

papapep514, 2022-12-10:
No words on this yet, patiently waiting...

Jacksonbm1, 2023-01-05:
Hi! Switchbot finally seems to have released the API: https://github.com/OpenWonderLabs/SwitchBotAPI-BLE/issues/5#issuecomment-1371725158

Is there any chance this will now be able to be incorporated?

Jacksonbm1, 2023-01-09:
Clarification, it's actually not an official documentation, someone just reverse engineered it, but that still seems good enough to use?

dsypniewski, 2023-01-09:
@Jacksonbm1 So the linked PR is from the official source but it only contains advertisement data structure, so you can only read some of the locks state.
What is available in the python library and HA as integration is based on me reverse engineering it, but this is still work in progress and I haven't yet made a PR to document it.

Jacksonbm1, 2023-01-10:
Got it, thanks for clarifying! Does the integration allow you to control via an esp32 or would you need to use the homeassistant hub directly?

dsypniewski, 2023-01-10:
@Jacksonbm1 If by esp32 you mean the Bluetooth proxy functionality, I haven't tested it myself but I heard from multiple sources that it works.

chinedu40, 2023-01-20:
> @Jacksonbm1 If by esp32 you mean the Bluetooth proxy functionality, I haven't tested it myself but I heard from multiple sources that it works.

As one of those people, i can confirm it works.

Jacksonbm1, 2023-02-20:
@chinedu40 I'm a little confused about how you did that. Would you by any chance be able to point me towards docs or even just your code?

BangerTech, 2023-03-10:
any news on the Lock integration?

chinedu40, 2023-03-30:
> @chinedu40 I'm a little confused about how you did that. Would you by any chance be able to point me towards docs or even just your code?

Apologies for the late reply. 

There's not special code needed....You need one of these devices from here: https://esphome.github.io/bluetooth-proxies/ . Once you have it, flash the software using the website. It will instruct you to add to Home assistant (you'll need to know the devices IP to add to home assistant). Once added to home assistant it will automatically find your switchbot bluetooth devices as long as you have the switchbot integration installed. Sometimes with the lock you may need to open and close it to refresh the Bluetooth connection and for it to be found on home assistant.  

Hope this helps!