devWaves/SwitchBot-MQTT-BLE-ESP32 #80: Won't connect after updating switchbot cuirtain firmware to 4.6

pranktap, 2022-06-25:
I updated my curtain firmware to v 4.6 and now this will not find my curtains. It was working fine before the uprade. Is this just me?

devWaves, 2022-06-26:
well crap....

For future reference, if it is working, I suggest not to upgrade the firmware. I will try to fix it though. I am not ready to upgrade my curtains yet if it has an issue, so I will need some help from you

it looks like the documentation for curtain has not been updated for firmware 4.6
github.com/OpenWonderLabs/SwitchBotAPI-BLE/blob/latest/devicetypes/curtain.md#0x45-curtain-setting-command
github.com/OpenWonderLabs/SwitchBotAPI-BLE/issues/4

a few questions/comments:

1. is it only control that is not working? if you control the curtain with your phone then reboot the esp32, are the values updated in mqtt/HA?
2. can you test sending a command using the nRF connect app? here is an example of how to send commands to the curtain https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/issues/48#issuecomment-1004152003
3. also using nRF connect app, can you provide some screenshots of what the the "Service Data" and what the UUID looks like. Perform a scan with nRF connect and find the device MAC address. Take a bunch of screenshots with the associated position % the curtain should be at. So do like 0% position, then scan in nRF connect and take a screenshot. Then 100%, then scan again in nRF connect and take a screenshot. Then 50%, then scan again in nRF connect and take a screenshot.

pranktap, 2022-06-26:
While troubleshooting the devices no longer show up in MQTT or home assistant anymore :( I’ll use the app tomorrow and get
You those values.

lst3245, 2022-06-26:
For 1), both the control and sensors are not working in HA. Sensors like battery and linkquality have no update since upgrading but illuminance and position seem to have some random garbage value updates (illuminance sometimes has unavailable value but stays at level 1, when it definitely is not 1 in the daytime, position seems to be jumping randomly)
[image]
[image]
[image]

For 2) and 3), I can control the curtain just fine using the 00, 32, 64 values for 0%, 50% and 100%. Here are the screenshots (although I am not sure if I have done it correctly):

00
[image]
[image]
[image]

64
[image]
[image]
[image]

32
[image]
[image]
[image]

devWaves, 2022-06-26:
@lst3245 thanks that is some good info. I think I found the problem. They have updated the UUID. It used to be 0x0D00, now it is 0xFD3D. It seems the newer devices and firmwares are being updated to use that value. Not sure why, but this should fix it...

in the code edit this line:
```if ((advertisedDevice->isAdvertisingService(NimBLEUUID("cba20d00-224d-11e6-9fb8-0002a5d5c51b"))) || isContactDevice(deviceName) || isMotionDevice(deviceName) || isMeterDevice(deviceName))```

to this...
```if ((advertisedDevice->isAdvertisingService(NimBLEUUID("cba20d00-224d-11e6-9fb8-0002a5d5c51b"))) || isContactDevice(deviceName) || isMotionDevice(deviceName) || isMeterDevice(deviceName) || isCurtainDevice(deviceName))```

the difference is the isCurtainDevice(deviceName) part added

I have the feeling all newer firmware updates will start using this UUID so I might just update it for the bot as well in anticipation

lst3245, 2022-06-26:
> @lst3245 thanks that is some good info. I think I found the problem. They have updated the UUID. It used to be 0x0D00, now it is 0xFD3D. It seems the newer devices and firmwares are being updated to [...]

Thanks so much. Just compiled the firmware again and it's working well in HA now. Much appreciated your effort :)

holdestmade, 2022-06-27:
> @lst3245 thanks that is some good info. I think I found the problem. They have updated the UUID. It used to be 0x0D00, now it is 0xFD3D. It seems the newer devices and firmwares are being updated to [...]

This worked for me too, many thanks

devWaves, 2022-06-29:
fix applied in v6.11

I applied the same style fix for the bot if they ever change the UUID

closed 2022-06-29 by devWaves

mike1627, 2022-07-02:
> in the code edit this line: `if ((advertisedDevice->isAdvertisingService(NimBLEUUID("cba20d00-224d-11e6-9fb8-0002a5d5c51b"))) || isContactDevice(deviceName) || isMotionDevice(deviceName) || isMeterDevice(deviceName))`

I just stumbled across this thread, I'm having the same issues here.  Which file in HA are you referring to that has this code?

devWaves, 2022-07-02:
@mike1627 hey there are no files to change in HA for this as it uses an ESP32. Firmware 4.6 starts using the new UUID = 0xFD3D = "0000fd3d-0000-1000-8000-00805f9b34fb", where before firmware 4.6 it used UUID = 0x0D00 = "cba20d00-224d-11e6-9fb8-0002a5d5c51b"

I am only given the option to upgrade to v4.4 at the moment, so it looks like v4.6 hasn't reached the Switchbot server I connect to. But others have confirmed that was the issue

@RenierM26 adding you so you are aware