devWaves/SwitchBot-MQTT-BLE-ESP32 #82: Switchbot (finger) not work after Update for Curtain firmware 4.6

chimcanhcut1112, 2022-06-30:
Hello, after update curtain work fine, but my switchbot not working,, not responding..

fantnhu, 2022-07-02:
Hi!
I confirm this as well. Switchbot curtain ROD 2, there are 2 of them.
The device information is received by the HA, but the control does not work. The log says it opens and closes.
What information do you need?
Thanks!
[image]

devWaves, 2022-07-02:
@fantnhu these are separate issues. User @chimcanhcut1112  is saying their curtain works but the bot is not working

@fantnhu the reason your curtain is most likely not working is because the esp32 is too far from the curtain. your linkquality rssi is -97 which is not good. try placing the esp32 closer to your curtain and try again

chimcanhcut1112, 2022-07-02:
> @fantnhu these are separate issues. User @chimcanhcut1112 is saying their curtain works but the bot is not working
> 
> @fantnhu the reason your curtain is most likely not working is because the esp32 is too far from the curtain. your linkquality rssi is -97 which is not good. try placing the esp32 closer to your curtain and try again
> 
> you are also using an esp32 that requires an antenna. make sure the antenna is attached correctly

i'm sorry for  mistake
I reinstalled it and it worked fine. (Previously I updated via bin file works only curtain)
Thank you

devWaves, 2022-07-03:
Just closing this as the initial issue mentioned is resolved

closed 2022-07-03 by devWaves