devWaves/SwitchBot-MQTT-BLE-ESP32 #88: bot is not recognized

GiladShoham, 2022-07-22:
I was able to flash everything correctly, and I see the hub in mqtt explorer (see image)
[image]

This is the value of the config message
```js
{
  "~": "switchbot/esp32_switch_bot_hub/esp32",
  "name": "esp32_switch_bot_hub Linkquality",
  "device": {
    "identifiers": [
      "switchbotesp_esp32_switch_bot_hub_MAC_ADD"
    ],
    "manufacturer": "WonderLabs SwitchBot",
    "model": "ESP32",
    "name": "esp32_switch_bot_hub"
  },
  "avty_t": "switchbot/esp32_switch_bot_hub/lastwill",
  "uniq_id": "switchbotesp_esp32_switch_bot_hub_MAC_ADD_linkquality",
  "stat_t": "~/rssi",
  "icon": "mdi:signal",
  "unit_of_meas": "rssi"
}
```

But I'm not able to recognize my bot.
Here is an image of the esp32 on both sides + the bot
In case it's hard to read, it says `ESP WROOM 32D`
[image]
[image]

Maybe it's my device that is not supported? or did I use the wrong version? or do I need to send some command to find the bot?

Thanks a lot!

devWaves, 2022-07-22:
I think someone mentioned using an ESP32 devkit v4 with ESP WROOM 32D but I would have to check through the HA forum chats

did you uncomment the part inside the bot list? the default code looks like this

```
static std::map<std::string, std::string> allBots = {
  /*{ "switchbotone", "xX:xX:xX:xX:xX:xX" },
    { "switchbottwo", "yY:yY:yY:yY:yY:yY" }*/
};
```

when you add a bot it should look like this with the inside uncommented

```
static std::map<std::string, std::string> allBots = {
  { "switchbotone", "xX:xX:xX:xX:xX:xX" }
};
```

GiladShoham, 2022-07-23:
No, I missed this part. I'll try it soon and will update here about it:)

GiladShoham, 2022-07-27:
@devWaves Works like a charm now
see the attached video - https://youtu.be/cxfHDoZxu54

Thanks for your help!

closed 2022-07-27 by GiladShoham