devWaves/SwitchBot-MQTT-BLE-ESP32 #90: Switchbots devices do not respond after 2 to 3 days

wellan35, 2022-08-04:
I have 2 switchbot curtains paired together, one switchbot bot and one contact sensor connected to my wemos di mini eps32. Like I said after a 2 to 3 days of working my contact sensor stops reporting to my home assistant. It (the switchbot devices) give an unavailable status or continues to report the same last state but do not record any new event. I have everything up to date with firmware 1.1, your code is in version v6.12. Maybe I have too many connected device?

This is the connected devices code:

/* Switchbot Bot Settings */
static std::map<std::string, std::string> allBots = {
  { "SwitchbotThermopompe", "C2:C2:78:DF:41:B4" }/*,
  { "switchbottwo", "yY:yY:yY:yY:yY:yY" }*/
};

/* Switchbot Curtain Settings */
static const int curtainClosedPosition = 10;    // When 2 curtains are controlled (left -> right and right -> left) it's possible one of the curtains pushes one of the switchbots more open. Change this value to set a position where a curtain is still considered closed
static std::map<std::string, std::string> allCurtains = {
  { "RideauxCuisine", "C6:BD:87:AE:EB:17" }/*,
    { "curtaintwo", "yY:yY:yY:yY:yY:yY" }*/
};

/* Switchbot Meter Settings */
static std::map<std::string, std::string> allMeters = {
  /*{ "meterone", "xX:xX:xX:xX:xX:xX" },
    { "metertwo", "yY:yY:yY:yY:yY:yY" }*/
};

/* Switchbot Contact Sensor Settings */
static std::map<std::string, std::string> allContactSensors = {
  { "PortePrincipale", "D4:FB:FD:48:2F:6F" }/*,
    { "contacttwo", "yY:yY:yY:yY:yY:yY" }*/
};

kenneyd, 2022-08-05:
I'm having the same issue, but on 6.11. After a couple of days, I noticed I stopped registering events from my door sensor and motion sensor in my MQTT server. I can still see the events if I use the switchbot app to view them (BLE direct, not using the hub).

Not sure if it's an overcrowding issue, since I currently have 12 devices.

wellan35, 2022-08-05:
> I'm having the same issue, but on 6.11. After a couple of days, I noticed I stopped registering events from my door sensor and motion sensor in my MQTT server. I can still see the events if I use [...]

It seems like an old bug, I also had this issue on 6.10. But I'm quite new to using this code so it was maybe in older version as well

devWaves, 2022-08-05:
Can you guys try using the arduino core ESP32 library v1.0.6 from the Boards manager libraries in arduino. I have a feeling v2+ has issues still related to BLE. v2+ added support for more esp32 boards but might have broke stuff for older boards. 

The issues may only arise with continuous scanning when contact sensor or motiom sensors are involved

wellan35, 2022-08-05:
Great to know, will try it out. Going on a week long road trip so won't be able to give feedback soon

1liminal1, 2022-08-07:
I'm also getting this. 

As an, admittedly convoluted, work around. I set up Node Red to poll the device and then use a smart plug to reset it when it becomes unavailable. This was working pretty well and resetting it once a day. 

I have switched to using the native Bluetooth integration, but its honestly much worse (I have a two bedroom apartment, its not huge, but I'm guessing radio congestion is a thing here). Its so slow and you cant set a device as a push button like you can with this software. Ill be switching back to this today sometime (can we get a better name then SwitchBot-MQTT-BLE-ESP32 HAHA?? Maybe ESPSwitchbot lol, something that rolls off the tongue a bit better lol) 

Also a little info about my environment. 

2X Curtains
3X SwitchBot
1X Thermomotor 

So not a lot of devices, and not spread over a very large area. 

Ill recompile with the Arduino libraries and see if I have a different experience. 

Has anyone else tried this yet?

devWaves, 2022-08-08:
@1liminal1 lol, the name will have to stick, as I wasn't really planning on originally making so many updates I just gave it a unique name that helped people find it when searching switchbot, esp and mqtt together

It is difficult for me to test all the libraries while also making code changes and figuring out where issues are coming from. The esp32 is basically being pushed to the limits with MQTT, webserver and BLE scanning all simultaneously. But I was able to reproduce some issue with ESP32 arduino core library v2+....just not sure yet what the exact issue is.  v1.0.6 has been solid for me for a long time though

I will have v7 out in a few days, just doing some final testing so I don't release something that is broken...but so far it is AWESOME!!  Mostly for the contact sensor/motion sensor performance, but many other improvements. Motion and Contact detection in v7 with multiple ESP32s is basically instant detection, dare I say as fast as zigbee/zwave

1liminal1, 2022-08-08:
> lol, the name will have to stick, as I wasn't really planning on originally making so many updates I just gave it a unique name that helped people find it when searching switchbot, esp and mqtt together

SwitchBot2MQTT?? ok, ill stop LOL

> It is difficult for me to test all the libraries while also making code changes and figuring out where issues are coming from

Totally understand and thank you, other then the crashing, this thing is awesome :) 

> dare I say as fast as zigbee/zwave

Thats a big call LOL ok, im really excited about this :) 

Thanks again

devWaves, 2022-08-08:
v7.0 ready for testing in the alpha build

https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/tree/v7alpha

***Still use v1.0.6 ESP32 arduino core library

kenneyd, 2022-08-08:
> v7.0 ready for testing in the alpha build

Thank you for this! I just downloaded the PlatformIO files in that branch, but I'm running into an error related to `isPlugDevice`. Should I be using the Arduino files for this alphav7 ?

```
src/SwitchBot-BLE2MQTT-ESP32.cpp:3905:167: error: 'isPlugDevice' was not declared in this scope
```

devWaves, 2022-08-08:
@kenneyd platformIO files are fixed now. But just note that my testing was done using the Arduino IDE and the ESP32 core v1.0.6 . I haven't extensively tested the latest ESP32 core files under PlatformIO

kenneyd, 2022-08-09:
I just installed the new version, will see if the problems re-surface. Thanks for the quick work 👍

devWaves, 2022-08-09:
v7.0 has been released to the main branch

1liminal1, 2022-08-10:
Hey, is there any info on the meshing? Is this just going to automatically happen, just install a couple of devices (for the size of your  apartment ect) and it will automatically create and manage the mesh right? 

Thanks for all your work on this, cant wait to get this installed :)

devWaves, 2022-08-10:
@1liminal1 meshing is not automatic. It needs to be configured. Setup one ESP32 and the primary ESP32, all other ESP32s are setup as secondary

Meshing is only beneficial to motion sensors, contact sensors and temp meters. Add the motion/contact/meter MAC address to all or most meshed ESP32s. Add the Bot and Curtain and Plug MACs only once on the ESP32 closest to the device

The readme is updated with instructions and there are example configurations in the Examples folder

1liminal1, 2022-08-10:
Awesome! Thanks for that info :) 

I guess I don't need it then really. I've just got bots and curtains. 

Does v7.0 speed up those devices, or its just the sensor devices? 

Thanks mate!

devWaves, 2022-08-10:
@1liminal1 v7 has some other improvements for bots and curtains but will not speed up the time it takes to send a command. Its the fastest it will go

1. curtain position slider will move as the curtain moves
2. bots/curtain will send a battery 1% if they return a weak battery response message when sending a command causing no action. This covers the scenario where the battery still shows high, but the device is unusable at that battery level
3. if you are using the switchbot hub/app and/or switchbot button alongside HA and you want HA to stay in immediate state sync, you can set alwaysActiveScan = true. This will technically use more battery on devices though (in theory). Otherwise the ESP32 will just scan on the configured schedules and after a command is sent
4. if the esp32 is already scanning and a state changed it will update HA

1liminal1, 2022-08-10:
Awesome! 

Thanks for letting us know. Seems like some cool features :)

kenneyd, 2022-08-12:
I still had problems on the v7 with platformio, so I moved to the arduino one and manually specified the 1.0.6 version for the esp32 library. 

I think the same could be done on platformio by pinning the `platform` version to `espressif32 @ ~3.5.0`, which looks like the last release running on 1.0.6, before they moved to arduino 2.0 in the [4.0 release ](https://github.com/platformio/platform-espressif32/releases).

I just made a PR to add this: https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/pull/93

1liminal1, 2022-08-16:
I have to say, its been 5 days now since compiling v7 with Arduino and it has been rock solid. No crashes, no issues. 

Id say if you're using platform IO... Dont lol

devWaves, 2022-08-22:
as @kenneyd mentioned v3.5.0 of the platform IO library uses the Arduino v1.0.6 which doesn't have all the issues from v2.0+ arduino.

v7.1 is released with the platform io libraries tested for 6 days. No issues to report

closed 2022-08-22 by devWaves