devWaves/SwitchBot-MQTT-BLE-ESP32 #92: Using ethernet instead of Wi-Fi (wESP32)

Eagleman7, 2022-08-10:
Thanks for building and maintaining this great piece of software!

My wESP32 has become quite unreliable for some reason. The latency for Wi-Fi is all over the place making it impossible to send commands to my Switchbot devices. This is probally unrelated to your code.

`
Reply from 192.168.3.16: bytes=32 time=956ms TTL=254
Request timed out.
Reply from 192.168.3.16: bytes=32 time=3564ms TTL=254
Request timed out.
Reply from 192.168.3.16: bytes=32 time=2160ms TTL=254
Reply from 192.168.3.16: bytes=32 time=46ms TTL=254
Reply from 192.168.3.16: bytes=32 time=603ms TTL=254
Request timed out.
Reply from 192.168.3.16: bytes=32 time=1206ms TTL=254
Request timed out.
Reply from 192.168.3.16: bytes=32 time=2148ms TTL=254
Reply from 192.168.3.16: bytes=32 time=1067ms TTL=254
Reply from 192.168.3.16: bytes=32 time=1879ms TTL=254
Reply from 192.168.3.16: bytes=32 time=230ms TTL=254
Reply from 192.168.3.16: bytes=32 time=266ms TTL=254
Reply from 192.168.3.16: bytes=32 time=148ms TTL=254
Reply from 192.168.3.16: bytes=32 time=148ms TTL=254
Reply from 192.168.3.16: bytes=32 time=345ms TTL=254
Reply from 192.168.3.16: bytes=32 time=235ms TTL=254
Reply from 192.168.3.16: bytes=32 time=203ms TTL=254
Request timed out.
Reply from 192.168.3.16: bytes=32 time=391ms TTL=254
Reply from 192.168.3.16: bytes=32 time=698ms TTL=254
Request timed out.
Reply from 192.168.3.16: bytes=32 time=177ms TTL=254
Reply from 192.168.3.16: bytes=32 time=63ms TTL=254
Reply from 192.168.3.16: bytes=32 time=339ms TTL=254
Reply from 192.168.3.16: bytes=32 time=17ms TTL=254
Request timed out.
Request timed out.
Reply from 192.168.3.16: bytes=32 time=727ms TTL=254
Reply from 192.168.3.16: bytes=32 time=2441ms TTL=254
Reply from 192.168.3.16: bytes=32 time=812ms TTL=254
Reply from 192.168.3.16: bytes=32 time=191ms TTL=254
Reply from 192.168.3.16: bytes=32 time=3067ms TTL=254
Request timed out.
Reply from 192.168.3.16: bytes=32 time=662ms TTL=254
Reply from 192.168.3.16: bytes=32 time=464ms TTL=254
Reply from 192.168.3.16: bytes=32 time=2762ms TTL=254
Reply from 192.168.3.16: bytes=32 time=7ms TTL=254
Reply from 192.168.3.16: bytes=32 time=114ms TTL=254
Reply from 192.168.3.16: bytes=32 time=235ms TTL=254
Reply from 192.168.3.16: bytes=32 time=411ms TTL=254
Reply from 192.168.3.16: bytes=32 time=417ms TTL=254
Reply from 192.168.3.16: bytes=32 time=48ms TTL=254
Request timed out.
Reply from 192.168.3.16: bytes=32 time=312ms TTL=254
Request timed out.
Reply from 192.168.3.16: bytes=32 time=194ms TTL=254
Reply from 192.168.3.16: bytes=32 time=408ms TTL=254
Reply from 192.168.3.16: bytes=32 time=265ms TTL=254
Reply from 192.168.3.16: bytes=32 time=116ms TTL=254
Request timed out.
Request timed out.
Reply from 192.168.3.16: bytes=32 time=397ms TTL=254
Request timed out.
Reply from 192.168.3.16: bytes=32 time=112ms TTL=254
Reply from 192.168.3.16: bytes=32 time=137ms TTL=254
Reply from 192.168.3.16: bytes=32 time=146ms TTL=254
Reply from 192.168.3.16: bytes=32 time=67ms TTL=254
Reply from 192.168.3.16: bytes=32 time=86ms TTL=254
Reply from 192.168.3.16: bytes=32 time=192ms TTL=254
Request timed out.
Request timed out.
Reply from 192.168.3.16: bytes=32 time=82ms TTL=254
Reply from 192.168.3.16: bytes=32 time=499ms TTL=254
Reply from 192.168.3.16: bytes=32 time=322ms TTL=254
`

I was wondering if it is possible to use the PoE ethernet port on my wESP32 instead of using Wi-Fi for network connectivity. Let me know if I can help you out with testing this!

devWaves, 2022-08-10:
do u specifically need the POE? if no they have esp32s with external antenna also that are cheap

as for the support of ethernet, you would want to check out this issue ticket reported in the EspMQTTClient library that I use...

https://github.com/plapointe6/EspMQTTClient/issues/97

are you noticing if the esp32 is rebooting at all? you would see a status "boot" message on the main MQTT topic. Just wondering if you're issue is more related to POE power supply issue

Eagleman7, 2022-08-17:
Sorry for the late reply, the device almost got fried because I left the window open and the wESP32 got soaked in water. Luckily it still works after drying for a day.

I thought POE would be helpful because then I can't have any issues with wireless connectivity and I don't need to run an external power supply near a wall outlet.

I cannot see anything boot related in MQTT:
[image]

Is the code able to connect through ethernet at all, currently it seems that it is just getting power through the ethernet cable but I do not see a DHCP lease.

devWaves, 2022-08-18:
no the code doesn't work with an ethernet cable. The library (espmqttclient) doesn't support non-wifi yet

Flachzange, 2023-08-20:
Hi all, I would like to resurrect the topic. I want to use my Olimex ESP32 POE and its Ethernet connection. I am actually super new to this. 

> no the code doesn't work with an ethernet cable. The library (espmqttclient) doesn't support non-wifi yet
Is that still the case? I least there is a constructor for MQTT only (without wifi)
I actually wanted to test this, but my bigger issue is actually that the sketch is getting too big in the moment I include the ETH.h and initiate the Ethernet connection.

@devWaves, is that realistic? Is the standard sketch already at its limits? Do you have a recommendation what I can safely remove, i.e. might be optional?

Flachzange, 2025-02-05:
nevermind, after 1,5 years with fresh eyes on it, I realised I can change the partition of the ESP32 to allow for more app space. Works now with Ethernet.