devWaves/SwitchBot-MQTT-BLE-ESP32 #94: [Enhancement] SwitchBot Strip Light Support

letnic, 2022-08-14:
Just add one of this node to my automation system and it's working great. 
Currently have a SwitchBot Strip Light and would really like to integrate it to my home assistant.

I try automation in Switchbot App but it's so slow, I can't make it fast enough to detect motion then open light. 
I think with this node I will be better.

I can help test and debug if necessary, if you have a way to decode BLE, I can try it.

Thanks you.

devWaves, 2022-08-14:
it is in the plans to be implemented when I get around to it. Cheers

letnic, 2022-08-18:
Is there anyways I can help you ?

devWaves, 2022-08-20:
@letnic not much you can do at the moment, but I'll let you know. Might not be for a bit. thx