devWaves/SwitchBot-MQTT-BLE-ESP32 #95: Long delay when turning light off immediatly after turning it on

1RandomDev, 2022-08-14:
Many times when I turn off a light immediately after it turned on, there's a really long delay until it turns back off. Don't know if this is on purpose or can be changed in the config, but I couldn't find any setting that solves it.

devWaves, 2022-08-14:
are u using the bot or plug mini?

the bot currently waits 2 seconds between every command sent to it using this value ```defaultBotWaitTime = 2``` . You can set that to 0 or 1 and it might help. it currently wait 2 secs between every command. This is because if the bot is in the middle of pressing and a command is sent to it, it will return a busy response. The code does retry on busy responses now though

consider if you had a hold value set on the bot of 5 seconds. You would want to wait at least 5 seconds before sending the next command otherwise you would get a ton of busy responses

when a command is sent the devices work by creating a connection, then disconnecting after a command is sent. This is to save battery

1RandomDev, 2022-08-25:
I'm using a SwitchBot on a light switch. And yes I have checked those 2 settings, but the delay in my case in from 20-60 seconds and is more or less completely random. Do you think it's more like a connection issue?

devWaves, 2022-08-26:
20-60 seconds would be a connection issue. how close is the bot to the esp32? if u place it real close does it have the same issue?

what is the typical rssi linkquality value foe the bot? you will want it at a.minimum in the 70s. 40-60 is good, 30-50 is excellent

Joe-DN, 2022-08-30:
I am having the same issue. Using a Switchbot on a light switch. Have a defaultBotWaitTime of 2. 

I only issue the command every 5 seconds. Works fine for 3-10 events, then the command just seems to get lost. After 20-60 seconds the command is received and triggers the switch bot. The ESP is 5cm from my switch bot so range isn't an issue.

Is there anything I can do to help debug?

devWaves, 2022-09-03:
I am unable to reproduce the issue. One thing to try if you haven't is to use an MQTT user/pass specific for the ESP32 alone, don't use the generic HA one. If using HA create a new account and use those credentials

In order to confirm the ESP32 is receiving the command to process it you will need to look at these things.

1) using MQTT explorer and the Arduino IDE you should see these appears in this order... (don't turn debugging on)

[image]
[image]
[image]
[image]