devWaves/SwitchBot-MQTT-BLE-ESP32 #97: lastwill offline

Capusjon, 2022-08-14:
Thanks for this project, it seems to be exactly what i'm looking for!
I ordered an ESP32 especially to try this out, it's my first go at using an esp32 so bear with me..

I'm beginning to think that mabey the problem is in my ESP but I'll lay it out anyway. 
Chip used = Esp-s3-12k https://www.waveshare.com/w/upload/6/6b/Esp-s3-12k_module_datasheet_v1.0.0.pdf

Whenever I upload the project, all seems to work fine for a couple of minutes, Haven't been able to move the curtains because it breaks before I can try.

[image]

this is what's posted on the mqtt broker, I would assume that this means the program is at least partially successful? Any Idea where I might look to get this running or is the ESP32 just not powerfull enough?

Kind regards

devWaves, 2022-08-14:
I can only guess atm that the issue is the ESP32-S3 chip which is a newer chip and may not work. I am not sure I use all ESP32-WROOM chips

Check here where it says the S3 is still not stable with arduino https://docs.espressif.com/projects/arduino-esp32/en/latest/getting_started.html#supported-soc-s

Capusjon, 2022-08-14:
Thanks for your fast response!
I ordered an ESP32 D1 Mini (ESP32-WROOM-32) should arrive next Wednesday to try this again.

I'll keep the S3 for future use of whatever might come up :) 

I'll close this topic for now since it doesn't seem to be an issue  we can fix here.

Thanks!

closed 2022-08-14 by Capusjon