devWaves/SwitchBot-MQTT-BLE-ESP32 #103: Enhancement: Switchbot bluetooth remote support

TTLucian, 2022-09-10:
https://eu.switch-bot.com/products/switchbot-remote

closed 2022-09-14 by devWaves

TTLucian, 2022-09-14:
Got it. Thanks! Made several requests to SwitchBot dev team to extend the remote functionality with the promise that they'll look into it. Here's hoping! The device has so much potential....

devWaves, 2022-09-14:
basically they just need to add a button push counter for up and down to the advertised data, the same way the contact sensor does it. Then it could be supported

you technically wouldnt want it linked to a device in that case though

evlo, 2022-11-30:
I think it needs firmware change as no packets seem to be sent on button press, but there is packet from the device, probably just saying it exist and battery is 0% or something like that
[image]

TTLucian, 2023-01-02:
Hi. New firmware was released for the remote some time ago. It seems to be cloud enabled now (through the Hub Mini) as it can be used to trigger scenes in the SwitchBot app and it no longer needs to be paired to only one device. Maybe it can be used now with your project.