devWaves/SwitchBot-MQTT-BLE-ESP32 #105: Occasional panics / reboots

TheHolyRoger, 2022-09-13:
I believe this is related to the NimBLE library as discussed here: https://github.com/h2zero/NimBLE-Arduino/issues/351

My Switchbot ESP32 devices occasionally panic and reboot.

I'm currently testing a fix here https://github.com/TheHolyRoger/SwitchBot-MQTT-BLE-ESP32/tree/feat/resetBluetoothStack

Resetting the bluetooth stack on a 5min interval seems to help the problem for me so far

devWaves, 2022-09-14:
hey @TheHolyRoger I am not experiencing thos issue using the ESP32 core library v1.0.6 which is what is recommended in the readme

are you using version greater than v2 by chance?

my esp32 hasnt missed a beat since I last updated it. I send myself a notification when the "boot" message is received and havent got one since the last time I rebooted the esp32

what esp32 dev board are u using?

TheHolyRoger, 2022-09-14:
I did notice the issue was tagged to v2.0+ but no I am definitely using 1.0.6 for the board and still experiencing these panic issues.

I'm using the DOIT devkit board, maybe I have a bad brand? Which are you using?

Arduino ESP32 Board library version 1.0.6
Board: DOIT ESP32 DEVKIT V1
Arduino 1.8.19
NimBLE-Arduino 1.4.0

devWaves, 2022-09-14:
I mainly use these 2 boards

ESP-WROOM-32-U (with external antenna)
https://a.aliexpress.com/_mMqQpko

Wemos D1 Mini
https://a.aliexpress.com/_m0lA00y

TheHolyRoger, 2022-09-14:
Thanks @devWaves I can't seem to find the same ones here in the UK but I'll try a different brand.

In the meantime have a look what I've done as a workaround:

https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/compare/main...TheHolyRoger:SwitchBot-MQTT-BLE-ESP32:feat/resetBluetoothStack

I've also lowered `rescanTime` in my config to 5 minutes, so every 5 minutes it will rescan and reset the stack before doing so. Seems to keep my units stable but not sure why you can't reproduce it!

TheHolyRoger, 2022-09-14:
[image]
Here is my "control group" unit without my fix applied, you can see how often it crashes

devWaves, 2022-10-01:
Hey I've been MIA for awhile, but my esp32s are still running 24/7 no reboots

You are running the native code only with config changes?

a rescan will do an active scan, so thats why I have the value high for rescanTime. Now that version 7 supports passive scanning. So setting it to 5 min will use more battery on devices

What kind of power supply are u using? use a very basic one or try while plugged into PC. Power supplies are known issues that ckukd cause reboots