devWaves/SwitchBot-MQTT-BLE-ESP32 #106: outcount not working

mousawm, 2022-09-19:
Hi, 

I'm using the latest release 7.1 but there is a problem with the outcount in switchbot contact sensor. I use it with D1 mini.

and I'd like to ask for feature if it possible to add contact sensor place ( is it placed inside the room or outside it ) so it corrects the incount and the outcount.

thanks

devWaves, 2022-10-04:
So I think thete is an issue with the actual contact sensor "exit" not working. You can test this by setting an automation in the switchbot app. Set an automation to "on contact sensor exit, open curtains" or "on contact sensor exit, press switchbot bot" ... if you also have a curtain or bot

it only seems to work when you press the contact button and then open the contact sensor (whcih equals exit). It doesn't seem to work when motion is detected and then the contact is open, which I believe it should

devWaves, 2022-10-09:
In order for exit to work on motion you need to change the "exit" setting within the Switchbot app. By default it is set to "press". For it to work on motion it needs to be set to "Motion". Works for me after changing that setting

I don't plan on making change to publish exit on entrance and entrance on exit. This can be done from within HA by creating a new entity

closed 2022-10-09 by devWaves