devWaves/SwitchBot-MQTT-BLE-ESP32 #109: SwitchBot Meter(s) - no update after a few minutes > Compile Error with older versions

FlorianRueb, 2022-10-26:
Hi!
I have the same issue as in thread #90 (starting directly after the first scan) but don't get it fixed! 
I tried to switch to the older version (Arduino ESP core 1.0.6) of the board infos and also downgraded to NimBLE Version 1.3.7, but now I get this compile error:
`More than one library found for "WiFi.h".
Used: /Users/flori/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exec: "python": executable file not found in $PATH
`
I am not sure how to get this working...

If i reinstall a newer version > 2.0, then compilation works fine again - but the problem isn't solved so far.
Do you have any hints for me?
Thanks!
Florian

FlorianRueb, 2022-10-26:
Got the compile error fixed in the meantime and compiled with Versions 1.0.6 and 1.3.7.

But unfortunately the problem is still there. About two minutes after startup (seems to be the setting of initialScan where I set up 120s) no other informations fly in.

I activated 'printSerialOutputForDebugging', and see a lot of messages like these:

> 00:39:24.559 -> START onResult
> 00:39:24.559 -> Advertised Device found: 
> 00:39:24.559 -> Name: , Address: fc:f1:8d:1b:13:3e, manufacturer data: 5900fcf18d1b133e
> 00:39:24.559 -> Adding Our Service ... 
> 00:39:24.559 -> metertwo
> 00:39:24.559 -> START callForInfoAdvDev
> 00:39:24.559 -> callForInfoAdvDev
> 00:39:24.559 -> START scanForeverEnded
> 00:39:24.640 -> forever scan Ended
> 00:39:24.640 -> END scanForeverEnded
> 00:39:24.640 -> END onResult

...but no information will be sent to the MQTT broker.

devWaves, 2022-10-27:
do you have only meters? During thr initial scan of 2 minutes it updates but not after that? even when rebooting the esp32 it only updates during the first 120 secs after boot?

FlorianRueb, 2022-10-27:
Yes, I only have a bunch of meters.
And yes, as you describe. Updates only during scan time.

devWaves, 2022-10-27:
Are your meters using the latest firmware? The latest code uses passive scanning which I don't think worked on older firmwares

The other variable used is defaultMeterMQTTUpdateSecs which forces an update every X seconds

If you hold the meter in your hand and the temp goes up it should update anytime the temp or humidity changes though

you can test withe settings alwaysMQTTUpdate = true. which will update mqtt every time a BLE message is received. If that doesnt work, then do alwaysActiveScan = true which will do active scanning instead of passive

FlorianRueb, 2022-10-28:
I think, they do. It's Version 2.6 and the App doesn't offer a newer one.

Yes, I tried with defaultMeterMQTTUpdateSecs = 60 with no success.
I also tried alwaysMQTTUpdate = true. Again no success. No value is sent, even if the Metered values change.
And a try for now is alwaysActiveScan = true. No success...

devWaves, 2022-10-29:
hey, so the latest firmware for meters available to me is v2.7

I just tested with one single meter and let it sit for a few hours and it had no noticeable issues. It was updating MQTT as expected. I didn't set any special settings, just the normal default configurations. Just to triple check, I put the meter in my hand for 5 seconds and MQTT updated with the new temp

my main suggestion is to figure out how to get firmware v2.7 because maybe v2.6 has those issues. Maybe connect to a VPN and get it that way? or contact switchbot and ask why you don't have access to v2.7

or maybe the issue is somewhere with MQTT. Try creating a unique user for the esp32.

if you haven't removed the batteries and put them back I would also suggest that, but may not fix the issue you are seeing. That fixed an issue for me a long time ago when a meter stopped broadcasting

FYI: v1.4.0 of NimBLE is no longer an issue with the latest code. Arduino Core v2+ is still an issue last time I checked so stick with v1.0.6

devWaves, 2022-11-01:
Ya my esp32 with only one meter in the device lists has been running for 2 days now. Still works and updates fine

Let me know if you can figure out how to get firmware v2.7 or if any of the other options work for you as it doesn't seem to be an issue with the code

FlorianRueb, 2022-11-01:
Yesterday I got version 2.7 installed (after asking two times for this particular update version), and since then it updates the values and polls out any change immediately (okay, it takes a few seconds, but I think that's due to the battery management).

So thanks for your support and for your great work!
Have a good day!

closed 2022-11-01 by devWaves

BantingBest, 2022-11-12:
Hello Florian, I'm also struggling with this problem.
How did you get the firmware?
And how did you flash them?
I can't find access to update 2.6 to 2.7.
I would be grateful for a hint.
Have a good weekend.

Nardol, 2022-11-12:
I was successfully able to update my meters after contacting Switch Bot asking specifically for 2.7 update for all my meters and after they replied to me I had the ability to update each meter in their own settings then Firmware and battery.

BantingBest, 2022-11-12:
Nardol, thank you very much for your fast reply!
I will contact SwitchBot via the feedback feature in the app.

BantingBest, 2022-11-13:
I received the update overnight.
Thank you for the quick support from the users of the project.
A big thanks to the owner of the resporitiy "devWaves" for taking the trouble to provide us with a perfect solution for SwitchBot.
Greetings, have a good time and stay healthy.

PS:
Please excuse my english, google helped me

FlorianRueb, 2022-11-13:
Hi Banting,

sorry for the late reply.
That’s true, it is not easy to get the new firmware for the switchbots. I wrote to them two times that I need an update from 2.6 to 2.7. The first time they told me that 2.6 is now downloadable for me (which I already had), the second time it worked!
Took a few minutes to get the updates work, and one switchbot needed a few reboots to reconnect to the system.

Best,
Florian