devWaves/SwitchBot-MQTT-BLE-ESP32 #111: Meters and bots register fine but Curtains are not appearing

RolandGoud, 2022-11-04:
First off....ty for all the work you have put in to this, normal BLE and the old integration was too slow for me. Hopefullly you can help me fix my last issues with the esp32 way of communicating with switchbot devices....

Just 'flashed' (is that the right term?) my first board ever....and i'm succesfull in getting a meter and a bot through the esp board. However i added the BLE mac addressess of my two curtains that are near the esp32. They are not registering, eventhough I am sure the mac addresses are correct.

Could it have something to do with passwords on these devices? And if so, do I need to reset the curtains to get rid of this or is there some other solution?

thanks in advance

devWaves, 2022-11-04:
My curtains are currently firmware v4.6 and I see there id a v6.0 out. I can't confirm the code works with v6.0. Is that what you have?

unless they added password support in v6.0, v4.6 and before does not support passwords for curtains

RolandGoud, 2022-11-04:
Yes V6.0 indeed

devWaves, 2022-11-04:
no v7alpha was my code. code is now v7.1 which is past that

so I'll need more details since I'm not quite ready to jump to v6.0 curtains if it totally breaks things. But I will update the code if easy

Do u see anything in HA? and u just can't control them? or in MQTT explorer?

If you can take a screenshot of the advertised service data with an app called "nrf connect" that will help me compare

bennyb0i, 2022-11-05:
I just [held my breath and] upgraded to v6.0 firmware for the curtains. Can happily confirm that v7.1 code still works without any apparent issue. Other than upgrading to v6.0 firmware, I didn't change any other settings for the curtains. I don't see any password feature for the curtains in the SB app, and it never prompted me when upgrading the firmware.

@RolandGoud, have you paired the curtains together in the SB app, or are using them individually? If you paired them, the pair will have its own BLE MAC address.

RolandGoud, 2022-11-06:
i havent paired them... I shte v7.1 code on production or is it in beta?

RolandGoud, 2022-11-06:
I haven't paired them... SO dont know what is wrong

RolandGoud, 2022-11-07:
I even did a full reset of one of the curtains to see what will happen...nothing

devWaves, 2022-11-07:
@RolandGoud are the curtains calibrated and working in the switchbot app? make sure nothing is connected to them, like close the switchbot app and turn off bluetooth on your phone

I'll look at upgrading mine to v6.0 but @bennyb0i already confirmed the code is compatible

neromaverick, 2022-11-08:
Not sure if this is related to this. But I've been having similar issues.

My switchbot works fine, however my contact sensor & curtain don't seem to be working properly.

From a clean slate, it takes quite a while for Home Assistant to populate the sensor & curtain. When it finally did, they appear to flap constantly between available and unavailable. I've unpaired them from the switchbot app, and anything else they may be paired to, wiped all the configs, etc.

It's pulled some sensor data off them, however I've never been able to trigger the curtains & I've never seen the contact sensor change it's state when I test it. The contact sensor I'm less worried about, as I'm mainly trying to get the curtain to work with HA.

I've got a V2 curtain btw, if that matters.

I enabled the serial debugger and this is what I see when I try to close/open the curtain.
```
02:06:28.150 -> CUSTOM notify service not found.
02:06:28.150 -> Sending failed. Disconnecting client
02:06:28.214 -> Sending command...
02:06:28.214 -> Attempt to send command. Not connecting. Try connecting...
02:06:28.214 -> Try to connect. Try a reconnect first...
02:06:29.115 -> Connected
02:06:29.147 -> Reconnected client
02:06:29.147 -> Connected to: 
02:06:29.147 -> <REDACTED MAC>
02:06:29.147 -> RSSI: 
02:06:29.147 -> -99
02:06:30.434 -> lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
02:06:30.434 -> CUSTOM notify service not found.
02:06:30.434 -> Sending failed. Disconnecting client
```
Let me know if there's anything else you need from me. Or if I'm being stupid and overlooking something totally obvious :rofl:

bennyb0i, 2022-11-08:
@neromaverick Your RSSI value is quite poor meaning that the signal is having a hard time reaching the ESP32 and may explain the intermittent connection and data issues. Have you tried moving the SB device or the ESP32 closer to one another?

Anecdotally I've found RSSI lower than -93ish results in intermittent data coming through.

Could also be a v2 curtain thing too, but sadly I only have v1 curtains to test with.

neromaverick, 2022-11-08:
They're literally touching, so I'd say they're pretty close :smile:

I am getting some good RSSI, but it's kind of a mixed bag, -31, -44, -108, -81. I tried a different ESP32 as well, since I got a 3 pack. Seeing the same results from that too.

I will note though, last night I was able to get it working to a degree. It was responding to the close/open commands from HA about 60% of the time. I restarted the ESP32 after that to see if that would improve things, but I haven't been able to get any commands working since.

It's pretty strange.

I guess a few questions, after a factory reset, is there any reason at all to connect it to the app? I would assume to calibrate it, right? If it's uncalibrated would this cause these issues, and if so, is there a way to calibrate it without using the app?

I want to avoid any possible interference. I'm not partial to the app at all, since I plan to use HA for all my integrations & control.

And if any devs want to use me as a guinea pig to test the v2 curtains, I'm more than happy to oblige :smiley:

neromaverick, 2022-11-09:
Here's the ones I got https://www.amazon.com/dp/B09C5RDZ8G (Teyleten Robot D1 Mini ESP32 ESP-WROOM-32). They seem to connect via Wifi without any issues. Also I don't seem to have any issues with my switchbot either, the switchbot has been much further away than the curtains even and it receives every command. One exception is if I send a command to the curtains then the switchbot won't trigger right away. I know that's very likely because the ESP32 is currently working out the command with the curtains, after it times out, the switchbot will trigger.

So I've followed these steps:
1. Remove the curtains from the app
2. Factory reset the curtains
3. Add the curtains back to the app & calibrate
4. Wipe all MQTT messages pertaining to the gateway from my broker
5. Power the ESP32
6. Wait for a few minutes for everything to sort itself out
7. Curtains appear on HA
8. Test closing/opening through HA

There's only been one moment I've been able to trigger the curtains, and even so it would trigger about 60% of the time.

While debug was on, I did notice a lot of discovery messages from all sorts of devices in my area (earbuds, laptops, speakers, keyboards, etc.). I'm wondering if there's just too much interference from other BT devices?

Kind of at a loss at the moment. I'll dig back into it tonight and continue working this all out. I'm not ruling out that I'm doing (or not doing) something simple that's throwing a wrench in this whole setup :rofl: 

I want to thank you all for your help on this. And @RolandGoud it's not my intention to hijack your issue, just looked like we were experiencing the same issues and hoping we can both get this settled. :smile:

codeinstitution, 2022-12-03:
I'm also running into this issue. I'm using a WROOM 32 board and the board is stuck in the "activescanning" mode. I have the curtain rod 2s on the version 6.0 firmware. I'm able to control everything just fine from my app, however, I can't even connect to the curtains via my ESP32. I'm not even getting the curtains to show up on MQTT Explorer--presumably it's not discovering them? I'm wondering if Switchbot changed something in the way BLE works with the new rod 2's.

I think this might be an issue withe the rod 2's.

codeinstitution, 2022-12-06:
> I'm also running into this issue. I'm using a WROOM 32 board and the board is stuck in the "activescanning" mode. I have the curtain rod 2s on the version 6.0 firmware. I'm able to control [...]

Figured out my issue. I originally bought https://www.amazon.com/dp/B0B19KRPRC **which do not work**

https://www.amazon.com/dp/B09C5RDZ8G worked with no issues.

Surpirsingly both register as ESP32-D0WD-V3