devWaves/SwitchBot-MQTT-BLE-ESP32 #112: wifi time out error

mikejrh, 2022-11-13:
I am getting this output when i run the sketch in Arduino-IDE on an Esp-Wroom board
"wifi:timeout when WiFi un-init, type=4".
The wifiscan sketch example works.

devWaves, 2022-11-14:
have you started with a fresh arduino install and cleared out the ~/Documents/Arduino/libraries folder ?

you could have a conflicting wifi library

what esp32 model are you using?

mikejrh, 2022-11-14:
using a ESP32-WROOM-32 board.
fresh install with these libraries: NimBLE-Arduino, EspMQTTClient.  ArduinoJson, CRC32, ArduinoQueue 
and I had to install  PubSubClient-2.8.0 after an error message in compilation.

devWaves, 2022-12-07:
not sure. I can't reproduce

Does you SSID or password has special or non-alphanumeric characters?

Tadeas3, 2022-12-08:
Both are alphanumeric only.